prusa3d/PrusaSlicer #10952: Prusa Slicer crashes on ASSERT when trying to remove a model

lorc, 2023-07-05:
### Description of the bug

When trying to remove a model from a plate, PrusaSlicer crashes.

This happens with any model.

There is the backtrace:

```
(gdb) bt

#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff41542d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff4104a08 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff40ed538 in __GI_abort () at abort.c:79
#4  0x00007ffff3cdd3b2 in std::__glibcxx_assert_fail(char const*, int, char const*, char const*)
    (file=file@entry=0x555556b389b8 "/usr/include/c++/13.1.1/bits/stl_vector.h", line=line@entry=1125, function=function@entry=0x555556b3fad8 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slic3r::ModelObject*; _Alloc = std::allocator<Slic3r::ModelObject*>; reference = Slic3r::ModelObject*&; "..., condition=condition@entry=0x555556b38076 "__n < this->size()") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/debug.cc:61
#5  0x000055555602f831 in std::vector<Slic3r::ModelObject*, std::allocator<Slic3r::ModelObject*> >::operator[](unsigned long) (__n=<optimized out>, this=<optimized out>)
    at /usr/include/c++/13.1.1/bits/stl_vector.h:1125
#6  Slic3r::GUI::ObjectList::object(int) const (this=0x55555d10f130, obj_idx=<optimized out>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectList.cpp:5043
#7  Slic3r::GUI::ObjectList::has_selected_cut_object() const (this=0x55555d10f130) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectList.cpp:2547
#8  0x0000555556065d53 in Slic3r::GUI::Plater::priv::can_mirror() const (this=<optimized out>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/Plater.cpp:4719
#9  Slic3r::GUI::Plater::can_mirror() const (this=<optimized out>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/Plater.cpp:7653
#10 Slic3r::GUI::ObjectManipulation::update_mirror_buttons_visibility() (this=0x55555d14a420)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectManipulation.cpp:813
#11 Slic3r::GUI::ObjectManipulation::update_if_dirty() (this=0x55555d14a420) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectManipulation.cpp:753
#12 0x00005555567e6b37 in Slic3r::GUI::ObjectManipulation::UpdateAndShow(bool) (show=true, this=0x55555d14a420)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectManipulation.cpp:518
#13 Slic3r::GUI::Plater::priv::update(unsigned int) [clone .constprop.0] (this=<optimized out>, flags=0)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/Plater.cpp:2296
#14 0x0000555555fbd7f0 in Slic3r::GUI::Plater::priv::delete_object_from_model(unsigned long) (this=0x55555d00a0c0, obj_idx=0)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/Plater.cpp:3059
#15 0x0000555556043038 in Slic3r::GUI::Plater::delete_object_from_model(unsigned long) (obj_idx=0, this=<optimized out>)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/Plater.cpp:6108
#16 Slic3r::GUI::ObjectList::del_object(int) (this=0x55555d10f130, obj_idx=0) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectList.cpp:1900
#17 Slic3r::GUI::ObjectList::delete_from_model_and_list(Slic3r::GUI::ItemType, int, int) (sub_obj_idx=-1, obj_idx=0, type=Slic3r::GUI::itObject, this=0x55555d10f130)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectList.cpp:3155
#18 operator() (item=..., __closure=<synthetic pointer>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectList.cpp:3331
#19 Slic3r::GUI::ObjectList::remove() (this=0x55555d10f130) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.6.0/src/slic3r/GUI/GUI_ObjectList.cpp:3356
#20 0x000055555676525f in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=<optimized out>, event=...)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/event.cpp:1431
#21 0x000055555676aee2 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (event=..., this=0x55555d10f130)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/event.cpp:1901
#22 wxEvtHandler::TryHereOnly(wxEvent&) (this=this@entry=0x55555d10f130, event=...)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/event.cpp:1624
#23 0x0000555556770c63 in wxEvtHandler::TryBeforeAndHere(wxEvent&) (event=..., this=0x55555d10f130)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/include/wx/event.h:4007
#24 wxEvtHandler::ProcessEventLocally(wxEvent&) (this=this@entry=0x55555d10f130, event=...)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/event.cpp:1561
#25 0x0000555556770d91 in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x55555d10f130, event=...)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/event.cpp:1534
#26 0x00005555565fe782 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (event=<optimized out>, this=<optimized out>)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/event.cpp:1650
#27 wxWindowBase::HandleWindowEvent(wxEvent&) const (this=<optimized out>, event=<optimized out>)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/common/wincmn.cpp:1553
#28 0x000055555667f884 in gtk_window_key_press_callback(GtkWidget*, GdkEventKey*, wxWindow*) (gdk_event=<optimized out>, win=0x55555d10f130)
    at /usr/src/debug/prusa-slicer/wxWidgets-78aa2dc0ea7ce99dc19adc1140f74c3e2e3f3a26/src/gtk/window.cpp:1323
#29 0x00007ffff488c34d in _gtk_marshal_BOOLEAN__BOXED
    (closure=0x55555cfb8410, return_value=0x7fffffff81b0, param_values=0x7fffffff8210, marshal_data=<optimized out>, invocation_hint=<optimized out>, n_param_values=<optimized out>)
    at gtk/gtkmarshalers.c:84
#30 0x00007ffff506f6c0 in g_closure_invoke (closure=0x55555cfb8410, return_value=0x7fffffff81b0, n_param_values=2, param_values=0x7fffffff8210, invocation_hint=0x7fffffff8190)
--Type <RET> for more, q to quit, c to continue without paging--
    at ../glib/gobject/gclosure.c:832
#31 0x00007ffff509d938 in signal_emit_unlocked_R.isra.0
    (node=<optimized out>, detail=detail@entry=0, instance=instance@entry=0x55555d117980, emission_return=emission_return@entry=0x7fffffff8320, instance_and_params=instance_and_params@entry=0x7fffffff8210) at ../glib/gobject/gsignal.c:3812
#32 0x00007ffff508e4cc in g_signal_emit_valist (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>, var_args=var_args@entry=0x7fffffff83d0)
    at ../glib/gobject/gsignal.c:3575
#33 0x00007ffff508ed34 in g_signal_emit (instance=instance@entry=0x55555d117980, signal_id=<optimized out>, detail=detail@entry=0) at ../glib/gobject/gsignal.c:3622
#34 0x00007ffff4b546e5 in gtk_widget_event_internal.part.0.lto_priv.0 (widget=0x55555d117980, event=0x5555576dfaf0) at ../gtk/gtk/gtkwidget.c:7812
#35 0x00007ffff4b639dc in gtk_window_propagate_key_event (window=0x5555575300b0, event=0x5555576dfaf0) at ../gtk/gtk/gtkwindow.c:8259
[... 43 lines cut ...]
2. Add any model
3. Delete it
4.  
 
```
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slic3r::ModelObject*; _Alloc = std::allocator<Slic3r::ModelObject*>; reference = Slic3r::ModelObject*&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    1684080 IOT instruction (core dumped)  prusa-slicer
```

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Arch Linux 

### Printer model

Voron 2.4 with Stealthburner

lukasmatena, 2023-07-10:
I am not able to reproduce this. I assume that you are running some build from the Arch repository. Are you able to reproduce the crash with the AppImage downloaded directly from our github ([here](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.6.0/PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220.AppImage))?

lorc, 2023-07-10:
Hi @lukasmatena,

Yes, I can't reproduce the issue with the AppImage as well so it must be some Arch build quirk. Sorry for the noise.

lukasmatena, 2023-07-10:
Ok, thanks for the update.

lukasmatena, 2023-07-11:
@lorc What package were you running? Was it from the official repo or from AUR?

lorc, 2023-07-11:
It is original repo:

```
$ pacman -Ss prusa
extra/prusa-slicer 2.6.0-3 [installed]
    G-code generator for 3D printers (Prusa fork of Slic3r)
extra/slicer-udev 2.6.0-3 [installed]
    G-code generator for 3D printers (Prusa fork of Slic3r)
```

```
$ pacman -Qi prusa-slicer
Name            : prusa-slicer
Version         : 2.6.0-3
Description     : G-code generator for 3D printers (Prusa fork of Slic3r)
Architecture    : x86_64
URL             : https://github.com/prusa3d/PrusaSlicer
Licenses        : AGPL3
Groups          : None
Provides        : None
Depends On      : boost-libs  curl  glew  mpfr  nlopt  tbb  qhull  openvdb  gtk3  libjpeg-turbo  opencascade
Optional Deps   : slicer-udev: 3D printer connection rules [installed]
Required By     : None
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 127.50 MiB
Packager        : Frederik Schwan <freswa@archlinux.org>
Build Date      : Thu 29 Jun 2023 01:57:40 AM EEST
Install Date    : Sat 01 Jul 2023 10:35:06 PM EEST
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : Signature
```

Rokin05, 2023-07-19:
Hi, idk how make a backtrace, i have the same problem with the Arch version 2.6 (2.6.0-3 or the last 2.6.0-4) : 

```
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slic3r::ModelObject*; _Alloc = std::allocator<Slic3r::ModelObject*>; reference = Slic3r::ModelObject*&; size_type = long unsigned int]: Assertion '__n <this->size()' failed.
Abandon (core dumped)
```

```
pacman -Qi prusa-slicer
Nom                      : prusa-slicer
Version                  : 2.6.0-4
Description              : G-code generator for 3D printers (Prusa fork of Slic3r)
Architecture             : x86_64
URL                      : https://github.com/prusa3d/PrusaSlicer
Licences                 : AGPL3
Groupes                  : --
Fournit                  : --
Dépend de                : gtk3
Dépendances opt.         : slicer-udev: 3D printer connection rules [installé]
Requis par               : --
Optionnel pour           : --
Est en conflit avec      : --
Remplace                 : --
Taille installée         : 166,10 MiB
Paqueteur                : Sven-Hendrik Haase <svenstaro@archlinux.org>
Compilé le               : mar. 18 juil. 2023 08:23:52
Installé le              : mer. 19 juil. 2023 10:03:47
Motif d’installation     : Explicitement installé
Script d’installation    : Non
Validé par               : Signature
```

```
Operating System: Arch Linux 
KDE Plasma Version: 5.27.6
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.4.3-zen1-2-zen (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i5-7440HQ CPU @ 2.80GHz
Memory: 15.5 Gio of RAM
Graphics Processor: Mesa Intel® HD Graphics 630
Manufacturer: Dell Inc.
Product Name: Latitude 5480
```

lukasmatena, 2023-07-20:
@Rokin05 @lorc I am sorry, but if you cannot reproduce the problem in our AppImage, we are not going to debug it.

gferon, 2023-07-30:
I was going to file a new issue, but I realized this was reported here. I can reproduce the issue with the Flatpak build as well.

Here's the stacktrace I got FWIW:

```gdb
Thread 1 "slic3r_main" received signal SIGABRT, Aborted.
0x00007ffff6a7149b in raise () from /usr/lib/x86_64-linux-gnu/libc.so.6
#0  0x00007ffff6a7149b in raise () at /usr/lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff6a5a857 in abort () at /usr/lib/x86_64-linux-gnu/libc.so.6
#2  0x0000555555a45d76 in std::__replacement_assert(char const*, int, char const*, char const*)
    (__file=__file@entry=0x555557c32010 "/usr/include/c++/11.3.0/bits/stl_vector.h", __line=__line@entry=1045, __function=__function@entry=0x555557c3bd00 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Slic3r::ModelObject*; _Alloc = std::allocator<Slic3r::ModelObject*>; std::vect"..., __condition=__condition@entry=0x555557c31ec3 "__n < this->size()")
    at /usr/include/c++/11.3.0/x86_64-unknown-linux-gnu/bits/c++config.h:514
#3  0x00005555565c97fb in std::vector<Slic3r::ModelObject*, std::allocator<Slic3r::ModelObject*> >::operator[](unsigned long) (this=<optimized out>, this=<optimized out>, __n=<optimized out>)
    at /usr/include/c++/11.3.0/bits/stl_vector.h:1045
#4  Slic3r::GUI::ObjectList::object(int) const (this=this@entry=0x55555dd0d3b0, obj_idx=<optimized out>) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectList.cpp:5043
#5  0x00005555565c98f6 in Slic3r::GUI::ObjectList::has_selected_cut_object() const (this=0x55555dd0d3b0) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectList.cpp:2547
#6  0x000055555651707c in Slic3r::GUI::Plater::priv::can_mirror() const (this=<optimized out>) at /run/build/PrusaSlicer/src/slic3r/GUI/Plater.cpp:4719
#7  0x00005555565195f0 in Slic3r::GUI::Plater::can_mirror() const (this=<optimized out>) at /run/build/PrusaSlicer/src/slic3r/GUI/Plater.cpp:7653
#8  0x00005555565fe4f8 in Slic3r::GUI::ObjectManipulation::update_mirror_buttons_visibility() (this=this@entry=0x55555dd3dbd0) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectManipulation.cpp:813
#9  0x0000555556603072 in Slic3r::GUI::ObjectManipulation::update_if_dirty() (this=0x55555dd3dbd0) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectManipulation.cpp:753
#10 0x0000555556603488 in Slic3r::GUI::ObjectManipulation::UpdateAndShow(bool) (this=0x55555dd3dbd0, show=<optimized out>) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectManipulation.cpp:518
#11 0x000055555653ab4b in Slic3r::GUI::Plater::priv::delete_object_from_model(unsigned long) (this=0x55555dc09ea0, obj_idx=<optimized out>) at /run/build/PrusaSlicer/src/slic3r/GUI/Plater.cpp:3059
#12 0x00005555565e3fa6 in Slic3r::GUI::ObjectList::delete_from_model_and_list(Slic3r::GUI::ItemType, int, int) (this=0x55555dd0d3b0, type=<optimized out>, obj_idx=0, sub_obj_idx=<optimized out>)
    at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectList.cpp:3155
#13 0x00005555565e4445 in operator() (item=..., __closure=<synthetic pointer>) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectList.cpp:3331
#14 Slic3r::GUI::ObjectList::remove() (this=0x55555dd0d3b0) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_ObjectList.cpp:3356
#15 0x0000555557b37d66 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=<optimized out>, entry=<optimized out>, event=..., handler=<optimized out>)
    at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1431
#16 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=<optimized out>, handler=<optimized out>, event=...)
    at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1406
#17 0x0000555557b38191 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (this=this@entry=0x55555dd0d3b0, event=...) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1901
#18 0x0000555557b382ed in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x55555dd0d3b0, event=...) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1624
#19 0x0000555557b383a0 in wxEvtHandler::TryBeforeAndHere(wxEvent&) (event=..., this=0x55555dd0d3b0) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:4007
#20 wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x55555dd0d3b0, event=...) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1561
#21 0x0000555557b3845a in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x55555dd0d3b0, event=...) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1534
#22 0x0000555557b38e24 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (this=<optimized out>, event=...) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1650
#23 0x000055555777b720 in wxWindowBase::HandleWindowEvent(wxEvent&) const (this=this@entry=0x55555dd0d3b0, event=...) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/wincmn.cpp:1553
#24 0x0000555557854ec8 in gtk_window_key_press_callback(GtkWidget*, GdkEventKey*, wxWindow*) (gdk_event=<optimized out>, win=0x55555dd0d3b0) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/window.cpp:1323
[... cut ...]
#51 0x0000555557a64ef7 in wxEntry(int&, wchar_t**) (argc=<optimized out>, argv=<optimized out>) at /run/build/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:503
#52 0x00005555564ac757 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /run/build/PrusaSlicer/src/slic3r/GUI/GUI_Init.cpp:61
#53 0x0000555555a3bff3 in Slic3r::CLI::run(int, char**) (this=<optimized out>, argc=<optimized out>, argv=<optimized out>) at /run/build/PrusaSlicer/src/PrusaSlicer.cpp:688
#54 0x00005555559fc398 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /run/build/PrusaSlicer/src/PrusaSlicer.cpp:947
```