prusa3d/PrusaSlicer #10953: Organic support ignores Top contact Z distance value = 0.1mm

bmiluch, 2023-07-05:
### Description of the bug

The organic support did not work well for my model. I tried to change the Z distance to 0.1 mm but it did not help. I noticed that for layer height 0.2mm, organic support and Top contact Z distance value 0.1mm both support layers and model layers are every 0.2 mm. That works as expected for grid support. At least some support layers start with height Z like e.g. 11.1mm when model layers have Z height e.g. 11.0  and 11.2.

### Project file & How to reproduce

[Flexi-Rex-improved.zip](https://github.com/prusa3d/PrusaSlicer/files/11962276/Flexi-Rex-improved.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-win64

### Operating system

Windows 10

### Printer model

Prusa MK3S

pauluchin, 2023-07-05:
I have same problem reply in Ender 6 and 3
[image]

rtyr, 2023-07-07:
Expected behavior. Organic supports does not support variable layer height, so contact z distance is rounded to multiples of main layer height.

It was mentioned in release notes, also a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9462

closed 2023-07-07 by rtyr