prusa3d/PrusaSlicer #10954: Extra extrusion following a color change deposits large blob on model

npburns224, 2023-07-06:
### Description of the bug

When initiating a color change mid-print, the printer will sometimes perform the following actions:
1. Pause printing
2. Move to park position
3. Initiate color change procedure
4. Prompt for user to confirm color change was successful
5. Extrude a small amount of filament while moving the nozzle back to where it left off
6. Lower the nozzle down to the model
7. **_Extrude a large blob of plastic onto the model_**
8. Continue the print job

Interestingly enough, this issue does **not** occur during every color change, however the following G-code is present in the places where it does: 
`G1 E4 F1500`

For example, here is a snippet of the G-code surrounding one of the problematic color changes:
```
; stop printing object Donut Cradle With Nameplate id:0 copy 0
;WIPE_START
G1 F8640
G1 X55.142 Y109.775 E-.35365
;WIPE_END
G1 E-3.64635 F2100
G1 Z1.1 F720
; Filament-specific end gcode
M600;
T0
M900 K0.0275 ; Filament gcode LA 1.5
; printing object Donut Cradle With Nameplate id:0 copy 0
G1 X53.411 Y97.325 F10800
G1 Z.8 F720
G1 E4 F1500
M204 S800
;TYPE:Perimeter
;WIDTH:0.649999
G1 F2700
```
You can see there is an unnecessary 4mm extrusion just before the printing resumes.

</br>

In my attempts to resolve the issue, I've discovered the following to be a reliable workaround:

  - Printer Settings > Extruder > Length (the second to last option on the page) > change from 4 to 0

After performing this step, the same line of G-code that was causing issue now reads `G1 E0.35 F1500`, which is the amount of retraction I have configured the slicer to use.

Here is the same section of G-code after performing the workaround:
```
; stop printing object Donut Cradle With Nameplate id:0 copy 0
G1 Z1.1 F720
; Filament-specific end gcode
M600;
T0
M900 K0.0275 ; Filament gcode LA 1.5
; printing object Donut Cradle With Nameplate id:0 copy 0
G1 E-.35 F2100
G1 X53.411 Y97.325 F10800
G1 Z.8 F720
G1 E.35 F1500
M204 S800
;TYPE:Perimeter
;WIDTH:0.649999
G1 F2700
```
You can see that the `G1 E4 F1500` line is gone. The printer is now using the proper retraction/deretraction values.

</br>

I have done quite a few manual color changes and have never experienced this issue until recently.

Again, this issue with the extra extrusion does not occur during every color change. In fact, the issue only seems to occur once in the project I uploaded.

### Project file & How to reproduce

[Donut Cradle With Nameplate (w: issue).3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11965402/Donut.Cradle.With.Nameplate.w.issue.3mf.zip)

**To Reproduce**
1. Open a multi-part model in Prusa Slicer, then adjust the following settings:
    - Enable Expert mode
    - Printer Settings > General > Extruders > set to at least 2
    - Printer Settings > Custom G-code > Tool change G-code > {if layer_z >= 0.2}M600{endif};
    - Print Settings > Multiple Extruders > Wipe Tower > disable
2. Assign each part of your model to one of the extruders.
3. Slice the model.
4. Inspect the G-code that occurs after color changes, or simply attempt to print the model.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

macOS 13.4.1

### Printer model

MK3S+

rtyr, 2023-07-25:
> You can see there is an unnecessary 4mm extrusion just before the printing resumes.
> 
> In my attempts to resolve the issue, I've discovered the following to be a reliable workaround:
> 
>     Printer Settings > Extruder > Length (the second to last option on the page) > change from 4 to 0

I would say this is the correct solution, not workaround. Those 4mm in "retraction when the tool is disabled" should not be set in this scenario, otherwise it will be used during the toolchanges. This setting did not work properly in previous PS versions, so this behavior did not occur.

rtyr, 2024-02-18:
Answered

closed 2024-02-18 by rtyr