prusa3d/PrusaSlicer #10955: Prusa Slicer 2.6.0 adding cracks

Fobozzz, 2023-07-06:
### Description of the bug

It appears Prusa Slicer 2.6.0. is adding "cracks" into the sliced assets.  2 different (from different sources) .STL files have been tested.  These files were obtained from Thingiverse, different users.   The cracks appear when slicing and not in any other 3D program.  Just when sliced and ready to print.

[image]
[image]

### Project file & How to reproduce

[Prusa-Cracks.zip](https://github.com/prusa3d/PrusaSlicer/files/11969111/Prusa-Cracks.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Original Prusa i3 MK3S

Fobozzz, 2023-07-06:
I did a few tests with different files by rotating them, but the "cracks" appear to be predictable but random places.  I did not notice the crack until the printed item displayed it.

mix579, 2023-07-06:
All I see in the photos is the regular Z seam? The seam is inevitable in FDM printing unless the object can be printed in case mode. You can control its location with the seam settings or using the paint on seam tool. For round shaped like the one in the photo the slicer can't hide in the corners like it can with angular objects.

sarusani, 2023-07-06:
As @mix579 mentioned, these are seams.
You can check out https://help.prusa3d.com/article/seam-position_151069 & https://help.prusa3d.com/article/seam-painting_168620 for additional information.

closed 2023-07-08 by rtyr

Fobozzz, 2023-07-10:
Wow - thank you very much.  I learn something new.   I had no idea about it
being a seam.

Thanks to all.