prusa3d/PrusaSlicer #10956: MMU2s and organic supports

JoHnEyAp, 2023-07-06:
### Description of the bug

when using the painting tool and support blockers for organic supports, the support blockers wont work.  If i remove the painting the support blockers work again.

the model i am printing has a logo and threads, printing the threads with supports will cause issues. 

Regular supports work fine.

### Project file & How to reproduce

1. choose organic supports
2. paint model or paint support blockers
3. slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

windows 10

### Printer model

mk3s+

Jan-Soustruznik, 2023-07-14:
SPE-1816

JoHnEyAp, 2023-07-14:
I can add files and a video if needed, but it was easy to reproduce.

Organic supports are awesome by the way

closed 2026-09-29 by prusaslicer-bot[bot]