prusa3d/PrusaSlicer #10959: 2.6.0 Bug when centering in object based cordinates

Plyn1x, 2023-07-08:
### Description of the bug

When typing cordinates in object based coridantes its behaing weird
[Körper168.zip](https://github.com/prusa3d/PrusaSlicer/files/11990644/Korper168.zip)

### Project file & How to reproduce

1. Imports STL
2. Add Modifier box etc.
3. then change to obejct based
4.  Type your postion and the problem should appear
5

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

Ratrig modified

Plyn1x, 2023-07-08:
https://youtu.be/eTHC2ioGQBY

Jan-Soustruznik, 2023-07-10:
Hello, @Plyn1x , 

Thank you for you report. Issue already exist. I closed issue as duplicate. 

https://github.com/prusa3d/PrusaSlicer/issues/10839

closed 2023-07-10 by Jan-Soustruznik