prusa3d/PrusaSlicer #10961: Synchronize skirt/shield with brim

vovodroid, 2023-07-08:
Currently  skirt/shield is printed independently from brim, causing shield instability. I would suggest whether bring brim to skirt or vise versa - skirt to brim. In the case of overhang goes out of brim brim should be extended. Other option is to print own skirt brim.

[image]

[image]

vovodroid, 2023-08-10:
> Other option is to print own skirt brim.

It could be helpful in such cases:

[image]

vovodroid, 2023-09-09:
Actually resolved by #11107

closed 2023-09-09 by vovodroid

kubispe1, 2024-03-05:
Hi @vovodroid, can you contact me to my mail petr.kubis@prusa3d.cz
Thanks and have a nice day.

vovodroid, 2024-03-05:
@kubispe1 , I've sent you email in the morning.