prusa3d/PrusaSlicer #10963: Strange slicing behavior, showing overhangs under the build surface

ianc1215, 2023-07-08:
### Description of the bug

When I try to slice certain objects the rendered preview shows non-existent overhangs on the model below the bed. 
I have already checked that the model is watertight and doesn't contain any errors.
I have also ensured the model is sitting correctly on top of the build surface and is not somehow sitting below.

[image]

### Project file & How to reproduce

To see the issue, just slice the model and examine the preview.
[Flexi-Rex-improved.zip](https://github.com/prusa3d/PrusaSlicer/files/11992384/Flexi-Rex-improved.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+win64

### Operating system

Windows 11

### Printer model

Modified Ender 3 Pro running Klipper 0.11.0-240

neophyl, 2023-07-08:
Those dark blue areas are bridges/overhangs.  If you check the legend on the print preview you will see that.  That makes narrowing the problem down much easier.  

You have your bridge flow ratio set to 95.  
[image]

1=100%, so you are using 950% flow on all your bridges.  Normally that value is set to something like 0.95 ie 95%
That looks like this
[image]

Jan-Soustruznik, 2023-07-11:
Hello, @ianc1215 , 

Thank you for your feedback. As @neophyl mentions, there is set incorrect Bridge flow ratio. Is used value out of range that cause mentioned issue. Do you remember how you set this value?

Thank you, have a nice day

ianc1215, 2023-07-11:
Honestly I don't recall but I have a hunch this is not a bug. I think I
might have tried using an older profile from a fork of Prusa Slicer (e.g.
SuperSlicer or something). Unless you can duplicate it I am pretty sure
this one is user error on my part.

neophyl, 2023-07-11:
That wouldn't surprise me, I've seen this very issue with the bridge flow ratio before when opening projects in older versions of PS and SuSi.  It is one reason I was able to identify it quickly.

closed 2026-09-29 by prusaslicer-bot[bot]