prusa3d/PrusaSlicer #10964: Organic support generation making sudden very large jumps of radius/position

davidmcnabnz, 2023-07-09:
### Description of the bug

When rendering organic supports, the position/radius of each trunk most of the time makes steady changes in radius and/or X-Y position. But often, the slicer generates the next support layer with massive jumps in radius/position. Sometimes, the radius increases by several millimetres, which leads to print failure of the affected supports.

### Project file & How to reproduce

In my case, I have used 'painted' supports, organic mode. I click on 'Slice Now', then export the GCode, then view it in the GCode Viewer app.

Slicing for eSun ABS, to print on Creality CR-10 Max.
[image]
[organic-support-jumps.zip](https://github.com/prusa3d/PrusaSlicer/files/11994071/organic-support-jumps.zip)

I've attached zip file containing project, rendered GCode and a screenshot showing the sudden radius jumps.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+linux-x64-GTK3

### Operating system

Ubuntu 22.04LTS

### Printer model

Creality CR-10 Max, factory firmware

Tupson444, 2023-07-09:
I had the same issue. I tweaked some settings in order to make the supports be regenerated until I got acceptable result (it was still shifted, but not too much so it was possible to print without failing).

davidmcnabnz, 2023-07-09:
My workaround is not completely pleasant -- remotely monitoring an HD cam feed, and watching for cracks or movement of support structures.

The instant any trouble appears, I pause the print, and carefully apply thick ABS slurry with a paintbrush on the outsides and insides of the trunks/branches, being careful to not get any on the top surface. This has rescued 2 3-day prints so far.

The Prusa Organic Supports system is amazing in so many ways. I really hope the Prusa devs can prioritise a fix for this issue, because it'll save a whole lot of failed prints.

murk-sy, 2023-07-10:
This is an issue of very optimistic angle settings and the slicer not really being able to process them. What your values are based on though I don't know.

Your settings:
[image]

Change based on Prusa Mini defaults
[image]

Using this there is no overhang issues on supports anymore.

The listed values are more wasteful but definitely reliable. I suggest you change to those and slowly modify them towards your current settings to suit your needs. I believe the main issue is the maximum branch angle, so maybe start with that at 45°.

davidmcnabnz, 2023-07-10:
Thanks @murk-sy for your quick investigation and reply.

The reason I've attempted 'optimistic' settings is to try to get the supports to 'snuggle closer' to the model mid-print as it starts to 'taper inwards'.

While the delamination issues persist on my printer, stand-alone support branches are vulnerable to layer separation, so getting them to 'interface' to the model rather than free-standing, would help to stabilise them. The mild scarring that results is a tiny price to pay, easy to file off, and will disappear entirely following acetone bath treatment.

When the current (3 day) print completes, I'll run a few test prints to explore the settings changes.

One of my greatest points of appreciation of Prusa's organic supports is being able to paint onto tall/narrow vertical structures, and see the slicer actually rendering effective support branches around them.

(With grid/snug support, this simply doesn't happen, and the slicer makes feeble token attempts of no benefit.)

Thanks again :)
David

Jan-Soustruznik, 2023-07-11:
Hello, @davidmcnabnz , 

thank you for you feedback. We know about this issue. We will try to fix it in some next release. 

I close issue as duplicate, https://github.com/prusa3d/PrusaSlicer/issues/10725

closed 2023-07-11 by Jan-Soustruznik