prusa3d/PrusaSlicer #10967: First purge location should be relative to print area.

VH-fixer, 2023-07-10:
Many of the MK4/XL videos showed prints with the first purge next to the print volume.  Not at the same default location that tends to wear a spot in print sheets currently.    There were even comments about this new function.   When PrusaSlicer 2.6.0 was released there was no option to control this.  Prusa Help states it is not a function in 2.6.0. 

The first purge of the nozzle on the print needs the option to automatically calculate a location relative to the print.   So it is not in the same location every new print.  

If  finding a default, clear location is too hard to code, give the user the option to select or drag-n-drop the location on the print sheet.   

Many users after seeing the demonstrations expected this ability to save their print sheets.  MK4/XL improvements have made it so Prusa 3D printing can move into the hands of more users.   Ones that want to slice, print and move on.  Editing G code to enable this function does not serve the current users or the new, less techie, users being added.

mix579, 2023-07-10:
The prime line location is defined in the startup GCODE, so you can define a custom printer profile and move it wherever you want to. 

I actually think it's a Good Thing they moved it back to the corner of the sheet. Originally I thought the location next to the object was cool and changed my Mk3S and Mini startup codes to mimic that behavior. But after a couple of weeks I realized that instead of wearing out just the left corner of the sheet that is not used for printing any real stuff anyway I now started to see "shadows" of the prime line in the middle of the sheet. So I reverted back to the default behavior for the Mk3, and it's interesting to see that the Mk4 default behavior was also changed. 

But again, if you're using the new variables introduced in PS 2.6, you can change the startup Gcode to put the line wherever you want.

rtyr, 2023-07-11:
It was like that initially, mainly because of variable mesh bed leveling size (that is likely the feature you mentioned).

Later we changed it, so purge line is always printed outside of print area. It prevents sheet damage in print area and also prevents getting the bed dirty when removing the purge by hand. New purge line is also easier to remove.

Firmware does few special probes just for that purge line, so it is printed correctly even if the main object is at different location.

closed 2023-07-11 by rtyr

nunofgs, 2023-07-20:
Thank you! I was going crazy searching the interwebs for why my newly-built MK4 was purging in the same corner as the MK3. I was sure I had seen videos of the new behavior.

This is definitely something that should be brought up in a blog post or something.

VH-fixer, 2023-07-21:
I understand putting the purge in the work area has been abandoned. Why not multiple purge locations out of the print area.  Nextruder only uses 1/4 of the sheet for the purge, not the full width as previous printers.   Allow 1, 2, 3 or 4 purge locations if it would fit.  Make it a selection during slicing for printers with the nextruder.   
This fix needs to be for everyone.  I can handle G-code in my sleep, but this is not a custom fix for me. This is to fix a long term problem for everyone.   Not just the few that can edit G-code.  
Just my thoughts on the matter, Thank You