prusa3d/PrusaSlicer #10969: Prusaslicer 2.6.0 invalid drive Z

boelle, 2023-07-11:
### Description of the bug

i get an invalid drive Z error on install

i expected a succesfull install

[image]

### Project file & How to reproduce

no project file just the PrusaSlicer_Win_standalone_2.6.0 installer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

does not matter 2.4

boelle, 2023-07-11:
ssame as here: https://github.com/prusa3d/PrusaSlicer/issues/2323

boelle, 2023-07-11:
just noted that the error comes when it wants to create shortcuts

[image]

but i de-selected everything, only prusaslicer and the required files are selected

[image]

[image]

rtyr, 2023-07-11:
@Janys

closed 2026-09-29 by prusaslicer-bot[bot]