prusa3d/PrusaSlicer #10977: Arrange Function

las-sir, 2023-07-12:
### Description of the bug

The Arrange function in 2.6.0 is affected by the modifiers (yellow). The Arrange spacing is applied to the modifiers since these are larger than the objects.

In 2.5.2 the spacing only applies in between objects regardless of modifier size.

If you arrange it correctly in 2.5.2, then opens it in 2.6.0 and perform slice as is, no error message pops up.
But if you perform arrange in 2.6.0 two of the objects wont fit the bed. Enable rotations are activated.

[image]
[image]

### Project file & How to reproduce

[BG MOUNTS.zip](https://github.com/prusa3d/PrusaSlicer/files/12026534/BG.MOUNTS.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

windows 10

### Printer model

MK4

Jan-Soustruznik, 2023-07-14:
Hello, @las-sir , 

thank you for your feedback. Yes, you have right, there is problem. We try to fix in some next release. 

Issue created: SPE-1690

lukasmatena, 2023-10-04:
This is already fixed in PrusaSlicer 2.6.1. Closing.

closed 2023-10-04 by lukasmatena