prusa3d/PrusaSlicer #10982: Auto-updater wrongly reports a newer version with lower version number

beeb, 2023-07-13:
### Description of the bug

The auto-updater in PrusaSlicer **2.6.0** reports that an update is available, and then proposes to update to version **2.5.2**.

Please note I had previously used the various beta and release candidate versions of 2.6.0.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11 Pro

### Printer model

N/A

rtyr, 2023-07-13:
@Janys

Janys, 2023-07-13:
This was caused by an old file cached on the update server. We have cleared the cache and the correct file should now be displayed and downloaded. Thank you for reporting this issue and apologies for the inconvenience!

closed 2023-07-13 by beeb