prusa3d/PrusaSlicer #10983: Top solid infill missing

iuliux, 2023-07-13:
### Description of the bug

All letters in this particular case were sliced correctly, except for one. I also have another similar piece that went well, but for this one the bug is reproducing every time.

[image]

### Project file & How to reproduce

[InspectionOnly.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12042588/InspectionOnly.3mf.zip)
My slicing settings should also be present in the 3MF file.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+MacOS-arm64

### Operating system

macOS 13.0, Apple M1 Pro

### Printer model

Mk3s+

neophyl, 2023-07-13:
Unfortunately it looks like your 3mf is not generated from PS as there are no settings stored in it.  If I load your 3mf in it uses the settings already loaded and it slices fine, for example -
[image]

To debug issues like this the prusa slicer project file is needed as the settings are as important as the model.

iuliux, 2023-07-17:
Did it load the color changes? I checked and those are in the zip.

Would settings in this format help?
[PrusaSlicer_config_bundle.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/12066882/PrusaSlicer_config_bundle.ini.zip)

Probably most impactful are the **0.6 nozzle** and:

[image]

iuliux, 2023-07-17:
Actually I've done some more experimentation and there's interesting behaviour when varying the Perimeter count

### Perimeters: 1

[image]

### Perimeters: 2

[image]

### Perimeters: 3 and above

[image]

Edu4444, 2023-08-17:
I have the same issue

Original .stl file:
[image]

Preview using 3 perimeters:
[image]

In my case top layer is generated when using 2 perimeters.

closed 2026-09-29 by prusaslicer-bot[bot]