prusa3d/PrusaSlicer #10987: Seam on vase mode

guistela, 2023-07-14:
### Description of the bug

when i slice a solid piece using vase mode it turns into a vase with seams!
[image]

### Project file & How to reproduce

Just slice any stl with vase mode enabled
[K3V2-Vase-3.3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12046270/K3V2-Vase-3.3.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 11

### Printer model

Ender 3 V2, Ender 5S1, Sovol 06 Plus

rainer1146, 2023-08-08:
[image]

Printed with Prusa Mk4. At the top with input shaper at the bottom with the normal profile.
Below you can see the seam a little. Absolutely unusable with input shaper.

C60nils, 2023-08-25:
I found when I commented out M74 W[extruded_weight_volume] in printer settings ->custom g-code in the before layer change section in prusa slicer the seam dissapeared.  Perhaps there is some bad/heavy maths happening with the weight calculation used in M74.

codefaux, 2024-02-02:
Issue still persists. Replicatable with a fresh install and builtin primitives.

Reproduction steps;
- Run ANY version, ANY profiles (No user config provided, cancel on setup wizard)
- Right-click, add primitive, Sphere
--- Yes I know a sphere would never print in Vase mode to begin with, it's the simplest model with non-perpendicular features. You can accomplish the same by rotating a cylinder by 25 degrees on the Y axis and cutting the top and bottom flat so it "makes sense" as a print if you need to justify it.

- Set Spiral Vase in Print Settings, click Yes to prompt
- Click Slice
[image]

[image]

[image]

[image]

Results;
Every layer, there is a Z-only move. No E specified -- not constant extrusion. It interrupts extrusion and moves purely veritcally, every layer. Any feature which is not purely perpendicular to the bed will cause a visual imperfection. Omitting this Z-only move fixes the issue. Here's a real-world impact on a regularly repeating model; a tall spiral can cooler sleeve. It's difficult to see so I've taken photos at vaguely different angles.

[image]

[image]

[image]

Present in 2.7.1 as well as several of the 2.6.x versions, I can't say exactly how far back this goes but it's before July '23.

closed 2026-09-29 by prusaslicer-bot[bot]