prusa3d/PrusaSlicer #10994: Prusa Slicer crashes when ejecting the SD card, Ubuntu

chrbau7, 2023-07-15:
### Description of the bug

 Prusa Slicer crashes, whenever ejecting the SD card after exporting. Happens since upgrading Prusa Slicer.

Log shows:
Error creating IO channel for /proc/self/mountinfo: Permission denied (g-file-error-quark, 2)
and
apparmor="DENIED" operation="dbus_bind"  bus="session" name="com.prusa3d.prusaslicer.InstanceCheck.Object14512492264124554885" mask="bind" pid=27856 label="snap.prusa-slicer.prusa-slicer"

### Project file & How to reproduce

[Room Name plate-letters_S_combined.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12060048/Room.Name.plate-letters_S_combined.3mf.zip)
This happens with every project.
Silice, export to SD card, eject SD card = crash.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+snap3

### Operating system

Ubuntu 20.04.06, Linux 5.15.0-76-generic x86_64

### Printer model

i3 MK3S

ivocavalcante, 2023-07-18:
Hi @chrbau7 ,

This error is related to the Snap package, not officially maintained by Prusa team. Could you, please, file a bug in Snap project [here](https://github.com/ivocavalcante/PrusaSlicer-snap) ?

chrbau7, 2023-07-24:
Thank you very much. That was indeed the problem. It would be great if the official Prusa App was available in the ubuntu snap store and/or at least the unofficial version was clearly marked as unofficial in the title. Even though it is written somewhere in the disclaimer, people would normally not read until there if they assume, it is the official Prusa Slicer, that they already know (ad if installed in the terminal that would not be seen at all). Thanks a lot.

closed 2023-07-24 by chrbau7