prusa3d/PrusaSlicer #10996: Very laggy while moving the object, viewport crashes

Stefankaa, 2023-07-15:
### Description of the bug

2.6.0 release is very laggy on my PC. It is clearly a bug made within the new update.
My PC specs:
- AMD Ryzen 9 7900X 12-Core Processor 4.70 GHz
- ASRock X670E Steel Legend
- NVIDIA GeForce RTX 3070
- 32GB DDR5 600MHz CL32 RAM
- Samsung 980 PRO 2TB M.2 SSD Drive
PrusaSlicer uses RTX 3070 graphics card, not the integrated one.

 ____________

**Please note that those bugs do not happen on 2.5.0 release!!! The 2.5.0 release works great, I've checked it again today and no issues at all.**
There are actually two bugs that I think are both related to some issues in viewport rendering:

### Low FPS, very laggy while moving the object
Moving the object causes lagging, there is a huge delay and like 15 fps. On previous slicer version everything goes smoothly. 
You can see how it looks like in the video I've attached. 
More instanced of an object cause more lagging. This is not related to objects size - objects with 20 faces also lag a lot.

### Importing big files causes viewport to crash
While importing big files (~1M faces) the viewport interface crashes and bed dissapears. Deleting the object causes the bed and interface to appear again. See the screenshots.

_______________

Things that I've already tried and they didn't help:
- updating RTX 3070 studio driver
- making sure PrusaSlicer uses RTX 3070
- reinstalling PrusaSlicer 2.6.0
- using standalone zip from github (PrusaSlicer-2.6.0+win64-202306191221_signed.zip)
- copying opengl32.dll to main slicer folder

### Project file & How to reproduce

### Low FPS, very laggy while moving the object
1. Add any object
2. Add a few instances
3. Move the object

### Importing big files causes viewport to crash
1. Add a big object (around 1M faces)
2. Viewport interface and bed disappears
3. Delete the object
4. Viewport interface and bed reappears again

[bug_report.zip](https://github.com/prusa3d/PrusaSlicer/files/12061661/bug_report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11 pro

### Printer model

Prusa i3 MK3s+, Prusa i3 MK3s

Stefankaa, 2023-08-05:
What about this issue? It's still lagging...

cupradito, 2023-09-18:
Hello.

Same here.

Prusa Slicer 2.5.0 Works fine.

Prusa Slicer 2.6.1 is extreme laggy, impossible to work with.... when I use the scroll on the right side to navigate across all elemens on the print area tooks 0.5 seconds per element.

Ryzen 5800X
16Gb Ram
Geforce GTX 1080Ti
Windows 10

Raystorm7, 2023-10-27:
Same here, bought new notebook with nve samsung 990 pro, Corsair Vengeance 32 gb ram, rtx 4060 8gb but the version 2.6.1 is very laggy, also the right scroll where are all stl models..
any help?

Stefankaa, 2023-10-27:
Seems that 2.7.0-alpha1 is ok regarding this issue. You should also try using this release and see if it works

krisrok, 2023-10-30:
> Same here, bought new notebook with nve samsung 990 pro, Corsair Vengeance 32 gb ram, rtx 4060 8gb but the version 2.6.1 is very laggy, also the right scroll where are all stl models.. any help?

I have the same issue as @Raystorm7 and it happens also on 2.7.0-alpha1. This seems to be related to rendering the model/parts list on the right side and is not neccessarily the same issue as @Stefankaa's.

Therefore I've just opened https://github.com/prusa3d/PrusaSlicer/issues/11557.

Stefankaa, 2023-11-01:
Hey guys. 
I think that it is the same issue as yours because I've checked it now and 2.6.1 also works ok somehow... So that makes sense if it happens also on 2.7.0-alpha1.

I have recently uninstalled tons of useless apps and background processes that come by default with Windows 11 (using ReviOS), so I think this is what has fixed this problem. I have also updated my graphics card driver in the meantime. Also VBS (virtualization based security) is turned off now. 

Maybe this issue is related to some background windows processes or the VBS itself? I'd rather guess that it is process-related conflict as I can see that some of you have Win 10. Maybe someone has used some of windows' resources in an incorrect way, which has lead to issues like that? Dunno.

Raystorm7, 2023-11-02:
> Seems that 2.7.0-alpha1 is ok regarding this issue. You should also try using this release and see if it works

After the update of 2.7.0-alpha1 the issue still persist.
The issue is the same as @krisrok as showed.
The notebook is a brand new one. I have installed the SO, drivers etc.. so I don't think there is some background process that can slow it :(

bombela, 2026-01-19:
Still the same issue for me as well since 2023. It seemingly gets even worse at ever new release. It sometimes takes 10s to scroll into the list of objects. Any UI panel is slow to scroll, making mistake very easy. Moving around object in the 3D view is insanity, because it stutters and lags so bad. Making you miss your target all the time. It also seems to lag in selecting/un-selecting object, and so you endup moving the wrong thing at the wrong place. In fact it will often highlight the old and new selection at the same time for a second, during which you never know which one is truly selected. In effect, handling more than 3 objects is impossible.

I tried bambulab studio (a fork of prusaslicer), and the 3D view was relatively smooth. A bit of stutter when changing selection but actually usable in contrary to prusaslicer. On the other hand, the rest of the UI of bambulab lags quite badly, but not as bad as prusaslicer. Loading files in bambulab studio is very slow, to the point where MS Windows thinks the application is dead. Prusaslicer loads files very fast in contrast.

This is on MS Windows 10 with an Nvidia RTX 3070. 4k resolution. I tried 8bit vs 10bit colors (HDR), 60hz vs 120hz, VRR off/on, mouse 125Hz vs 1kHz. No difference.

On a linux laptop, 1080p, 8 bit, integrated GPU, prusaslicer is much faster. The same linux laptop connected to the screen above at 120Hz 8bit doesn't lag.

So maybe it is specifically on MS Windows that it lags so bad.

closed 2026-09-29 by prusaslicer-bot[bot]