prusa3d/PrusaSlicer #11000: V2.6 crashes when going back to 3D edit mode from preview (WIN 11)

hannes-m, 2023-07-16:
### Description of the bug

after slicing the Version 2.6 crash when clicking on 3D edit mode

### Project file & How to reproduce

Cannot save it after crash :-)
(It happens every time going back after slicing)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

Win11

### Printer model

i3 MK3S

Jan-Soustruznik, 2023-07-17:
Hello, @hannes-m , We cannot reproduce your reported issue. Could you share some model(stl,3mf) whitch you had issue. 

Thank you, have a nice day

hannes-m, 2023-07-17:
Hi Jan,
I can reproduce it the following way:
- Import STL
- Slice it
- export the file to an external SD card
- Eject the SD card with the button in Prusa Slicer
- go back to 3d edit mode will crash the application then

Without the procedure with the external SD card I cannot reproduce the failure too.
Hope this helps...Hannes
[Akkuhalter_hinten v2.zip](https://github.com/prusa3d/PrusaSlicer/files/12067001/Akkuhalter_hinten.v2.zip)

hannes-m, 2023-07-17:
I tried an other external SD card reader: same behaviour, application crashes...
Attached file: SysInfo of my PC
[image]

closed 2026-09-29 by prusaslicer-bot[bot]