prusa3d/PrusaSlicer #11001: Mirror cutting

MxTale, 2023-07-16:
### Description of the bug

If you mirror an object (on X axis for example) then perform a cut of the mirrored model (adding connectors) the cut model will not be the mirrored one but the original one

### Project file & How to reproduce

[Piece 03 DAN V11.zip](https://github.com/prusa3d/PrusaSlicer/files/12064748/Piece.03.DAN.V11.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

W11

### Printer model

Anycubic Kobra

YuSanka, 2023-07-19:
Thanks for your report.
We are working on it (SPE-1819)

lukasmatena, 2023-11-15:
This is fixed since [2.6.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1). Closing.

closed 2023-11-15 by lukasmatena