prusa3d/PrusaSlicer #11005: options to control organic support rafts

Commod0re, 2023-07-17:
**Is your feature request related to a problem? Please describe.**
Sometimes the tiny islands are just not enough contact area on the bed. It would be nice if there were more options to control this

**Describe the solution you'd like**
additional options for organic supports:
- ability to either control the raft size, or ability to attach brims to them (currently in 2.6.0 brims do not attach to organic support islands)
- ability to either print organic supports on a raft, or ability to connect all the islands together into a single support raft

rtyr, 2023-07-17:
Already implemented. "First layer expansion" setting.

closed 2023-07-17 by rtyr

Commod0re, 2023-07-17:
You didn't even read my request, that does not do what I want

Commod0re, 2023-07-17:
thanks for closing my issue without bothering to even comprehend it -_-

rtyr, 2023-07-17:
It does what you wanted. You can specify the raft size for organic supports...

rtyr, 2023-07-17:
> ability to either control the raft size

Some screenshots for you... Please do not open other duplicates.

[image]
[image]

rtyr, 2023-07-17:
> thanks for closing my issue without bothering to even comprehend it -_-

It seems you started to rant without bothering to try it ;-).

Commod0re, 2023-07-17:
that's only one of the things I asked for though...

Commod0re, 2023-07-17:
you didn't even really answer my question in a helpful way, you just said it was already fixed and you didn't elaborate on how until I had opened multiple duplicates, and you continued to prove that you only read half of my request

Commod0re, 2023-07-17:
does that technically work? yes
is it the most wasteful version of the answer possible? also yes

whatever, good enough I guess