prusa3d/PrusaSlicer #11008: Nozzle diameter in Printersettings?

tom-321, 2023-07-17:
I'm not sure if this is a feature request or just an question how to do this:

I've changed my nozzle to 1mm and realized that perimeters are printed way to fast, 
but my profile for the 1mm nozzle is located in printERsettings and not printsettings ?!

Doesn't it make more sense to move nozzle diameter setting to printsettings 
or just link it in some way to overwrite the speed settings ?

mich-j, 2023-07-19:
Your profile in Print Settings probably have extrusion width set explicitly. If you go to Print Settings in Advanced mode, then to Advanced tab, you will see extrusion width settings. If those settings are set to 0, Prusa Slicer will compute them basing on nozzle diameter set in Printer Settings. 

Take a look at attached screenshot. This is the message that you will get if you hover your mouse over Perimeters option in Extrusion width section.

[image]

tom-321, 2023-07-19:
Hi, thanks, 
I know that, but my example was about speed, not width

neophyl, 2023-07-20:
Set the maximum volumetric flow in the print settings and that will act as a limit on your speeds.

tom-321, 2023-07-20:
Hi, thanks for the tips concerning the speed setting,
but this speed thing was just an e.g. to show the issue between the different setting profiles.
I need to create every time a new physical machine and new printERsettings preset and printsettings and set dependencies ... just to change nozzle diameter

lunatic1972, 2023-07-20:
Actually if you look under Auto Speed, look for Max Volumetric Speed, this will allow you to not have to adjust your speeds manually as the speed will never be greater than max. Well, that's not exactly true because there are certain instances but it will slow the speed down to be "close" to the max. Sometimes going over by about 1-2mm3/s. Basically even if you told it to go 300mm/s it would never reach that being maxed out at that arbitrary value. Mine is set at 14.1. I have a dragon standard flow which is around 17mm3/s.

closed 2026-09-29 by prusaslicer-bot[bot]