prusa3d/PrusaSlicer #11011: options to control organic support rafts

Commod0re, 2023-07-17:
posting again because someone closed my original request without bothering to comprehend it even a little, and I can't re-open it myself.

**Is your feature request related to a problem? Please describe.**
Sometimes the tiny islands are just not enough contact area on the bed. It would be nice if there were more options to control this, and NO, the **First layer expansion** setting does not address this!

**Describe the solution you'd like**
additional options for organic supports:
- ability to either control the raft size, or ability to attach brims to them (currently in 2.6.0 brims do not attach to organic support islands)
- ability to either print organic supports on a raft, or ability to connect all the islands together into a single support raft

rtyr, 2023-07-17:
Already answered in https://github.com/prusa3d/PrusaSlicer/issues/11005. closing

Commod0re, 2023-07-17:
answered incorrectly though!

rtyr, 2023-07-17:
correctly