prusa3d/PrusaSlicer #11013: Regarding the request to add X/Y axis acceleration control options for the I3 model. 关于请求增加I3机型的X/Y轴加速度控制选项

OuDCan, 2023-07-18:
The I3 structure 3D printer completes the Y-axis movement by moving the printing platform back and forth. It has a larger weight relative to the X-axis movement element, and as the printing work progresses, the load on the Y-axis movement becomes larger, and the Y-axis acceleration is limited lower than the X-axis acceleration. Therefore, in order to improve the overall printing speed of the printed parts while ensuring that the quality of the printed parts is good enough, I request to add X/Y axis acceleration control options for the I3 model. Thank you!

I3结构的3D打印机，是靠打印平台前后运动完成Y轴的运动的，它相对于X轴运动的元件，有着更大的重量，并且随着打印工作的进行，Y轴运动的负重也就更大，Y轴加速度相对于X轴加速度也就被限制得更低。所以，为了在保证打印件质量足够好的前提下提高打印件整体打印速度，我请求增加I3机型的X/Y轴加速度控制选项，谢谢！

Tupson444, 2023-07-18:
One way to do this is to use custom G-code at certain layers (right-click on the "+" and choose "add custom G-code"):

[image]

If your printer's regular acceleration is e.g. 2000, you can put `M201 Y1500` at say layer 200, then `M201 Y1000` at layer 400, etc. which would reduce printer acceleration on Y axis only.

But I agree that this would be better if done automatically, based on the calculated weight of the printed part (with thresholds to lower acceleration when a certain weight is reached).

rtyr, 2023-07-19:
> But I agree that this would be better if done automatically, based on the calculated weight of the printed part (with thresholds to lower acceleration when a certain weight is reached).

This is already possible in 2.6.0
You can for example add this line to "before layer g-code" g-code section:
`M201 Y{interpolate_table(extruded_weight_total, (0,2000), (300,1500), (2000,800))}`

It will start at M201 Y2000, then interpolated after each layer to 1500 till 300g, then interpolated to 800 till 2000g.

OuDCan, 2023-07-21:
I think this is a great feature update! 
But in order to get better surface effects on the print, the values for the speed and acceleration of the outer contour are usually set lower to achieve more ideal results. If only the threshold is limited, I think that if the speed and acceleration values for the outer contour printing are the same, increasing the speed and acceleration of the internal printing will save more printing time. 
So I suggest differentiating the settings for X/Y axis movement instead of just limiting the threshold.

我想这是个非常棒的功能更新！
但是为了得到更好的打印件表面效果，通常都会将外部轮廓的速度和加速度的数值设置得更低，以达到更加理想的效果，如果只是对阈值做限制，我想，“外部轮廓打印速度和加速度值相同的同时，将内部打印速度和加速度提高”这样的设置应该会节约更多的打印时间。
所以我建议对X/Y轴的运动设置做区分，而不是只对阈值做限制。

closed 2026-09-29 by prusaslicer-bot[bot]