prusa3d/PrusaSlicer #11016: Support interface leaves extra gap below perimeters

jamespearson04, 2023-07-18:
### Description of the bug

When attempting to support models with floating perimeters, support interface leaves an extra layer's gap below the perimeters to be supported, unless support distance is set to 0. For example, 0.2mm gap will leave a 0.2mm gap below bottom layer areas, but a 0.4mm gap (for a 0.2mm layer height) below the perimeters of the floating section, as demonstrated in the attached images.

This renders the support very ineffective, as either the distance to the perimeter is too high, or the distance to the bottom is too low.

[image]
[image]
[image]

### Project file & How to reproduce

Slice this, or any other model with floating perimeters, using support with any level of interface enabled. Observe the top layers of support interface - the last layer should be missing material below the perimeters of the supported part.

[Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/12089240/Issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.4 through 2.6.0

### Operating system

Windows 10 Version 10.0.19044

### Printer model

Ender 3, Ender 3 Max, Custom Replicator Clone

rtyr, 2023-07-19:
It's correct behavior with your settings. You have "Think bridges" enabled, therefore those blue perimeters uses thick bridge flow based on the nozzle diameter (0.4mm round extrusion - so it needs extra space below). You will likely get better results in general if you disable that "thick bridges" option.

jamespearson04, 2023-07-19:
Aha, many thanks! That sorts my issue, though it seems a bit of a shame that Thick Bridges and Supports are essentially mutually exclusive.

closed 2023-07-19 by jamespearson04