prusa3d/PrusaSlicer #11018: Still can not run 2.6.0 on Raspberry Pi 4B (ARMV7L)

daizhuo, 2023-07-19:
### Description of the bug

I can run this version:
PrusaSlicer-2.4.2+linux-armv7l-GTK2-202204251109

Any version **newer than** 2.4.2 **CAN NOT** run.

I do not run 64 bit system, so can not run AARCH64 slicer.

My operation system info:
Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

I'm really want to run 2.6.0 on my Raspberry Pi, I love 2.6.0 version!
Thanks a lot! My dear guys!

### Project file & How to reproduce

Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

### Printer model

PRUSA MK3S+

Jan-Soustruznik, 2023-07-24:
Hello, @daizhuo , 

Could you share what OS use - link to download if possible. Thank you

daizhuo, 2023-07-25:
> Hello, @daizhuo ,
> 
> Could you share what OS use - link to download if possible. Thank you

Hi.......The infomation from my rasberrypi is in the below.
I'm a new bird in Raspberry Pi, I just know I didn't install 64 Bit OS.
I downloaded the OS from this site:
https://www.raspberrypi.com/software/

I hope these information is helpfull for you to inspect the issue.
Thank you so much for your help!

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ uname -mrs
Linux 5.10.103-v7l+ armv7l
pi@raspberrypi:~ $

closed 2026-09-29 by prusaslicer-bot[bot]