prusa3d/PrusaSlicer #11021: Klipper Gcode flavor + Toolchange = SET_PRESSURE_ADVANCE inserted.

viesturz, 2023-07-19:
### Description of the bug

Prusaslicer inserts unexpected and unremovable SET_PRESSURE_ADVANCE ADVANCE=0 comands
when unloading filament for toolchange and never restores it.

Klipper saves pressure advance change for the current extruder and never restores the original advance.

### Project file & How to reproduce

[Base.zip](https://github.com/prusa3d/PrusaSlicer/files/12099274/Base.zip)
Set gcode flavor to Klipper
Add more than 1 extruder, slice something with multiple colours.

Reproducible even if retraction length on toolchange is set to 0.

Recommended change: remove the command. Klipper tool changes are generally gcode scripts and expecting any tuning like this to happen there, with slicer focusing on actual slicing.

Relevant Gcode snippet:
```
; CP TOOLCHANGE UNLOAD
;WIDTH:0.5
SET_PRESSURE_ADVANCE ADVANCE=0
;WIDTH:0.5
G4 S0
G1 E-5 F2400
;WIPE_START
G1 F14400
G1 X201.281 Y179.446 E-.16653
;WIPE_END
G1 E-4.83347 F2400
G1 Z4.1 F18000
M104 S245 T0 ; set temperature ;cooldown
M104 S250 T1 ; set new tool temperature so it can start heating while changing
T1
M109 S250 T1 ; set temperature and wait for it to be reached

```

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Win11

### Printer model

Voron 2.4

Bumblebiber, 2023-07-28:
My workaround is to use GCode substitution and replace "SET_PRESSURE_ADVANCE ADVANCE=0" by "SET_PRESSURE_ADVANCE ADVANCE=0.053"

viesturz, 2023-07-28:
🤦

bulcke, 2023-12-23:
This bug is still present in 2.7.1

bulcke, 2023-12-23:
> My workaround is to use GCode substitution and replace "SET_PRESSURE_ADVANCE ADVANCE=0" by "SET_PRESSURE_ADVANCE ADVANCE=0.053"

Or replace "SET_PRESSURE_ADVANCE ADVANCE=0" with nothing at all. The gcode line will be blank and PA in klipper will not be changed at all.

gluap, 2026-09-02:
Leaving comment as I'm still hoping for pressure advance to work with multiple extruders.