prusa3d/PrusaSlicer #11023: Add Electricity Costs

LukasQ, 2023-07-20:
Hi there

[User-Story]
As a maker, I want to have an estimation, how much my prints cost so I could either calculate an rough selling price or tell my friends the production costs in the could-you-do-me-a-favour prints. This is already possible through the filament price calculation, but running costs like energy are missing. 
Please add an input field where I could insert my current energy costs (in [€/kWh]) and add appropriate printer energy consumption profiles (or an additional input field).

**It would work like:**
_runtime of the print * printer consumption per hour * energy costs_
This still doesn't provide TCO, but is a good approximation. 

**Alternative:**
It still could be measured through a power meter and estimated manually, but the cost functionality is already implemented in Prusa Slicer

rtyr, 2023-07-20:
You may try this https://blog.prusa3d.com/3d-printing-price-calculator_38905/

neophyl, 2023-07-20:
Just add print time to your gcode output name and then work out how much power your print uses per hour using a suitable monitoring device.  Simple math at that point as the time is part of your file name.  

You are never going to get it real detailed without a realtime electricity monitor anyway so all you need is a rough estimate to base your costings on.  Print time is as good as anything else.

Also this issue is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8531

LukasQ, 2023-07-20:
Whops, overlooked that issue.

closed 2023-07-20 by LukasQ