prusa3d/PrusaSlicer #11037: Prusaslicer sending to Mini+ with PrusaLink: always succeed only on second attempt

doegox, 2023-07-24:
### Description of the bug

Hello

I followed https://help.prusa3d.com/guide/wi-fi-and-prusalink-setup-mini-_316781 to install Wi-Fi on my Mini+.
When trying to send a file from Prusaslicer v2.6.0, it systematically times out on the first try and systematically succeeds on second try. So maybe there is something doable to make Prusaslicer try harder and and avoid the first fail.

### Project file & How to reproduce

* Mini+ being turned on since a while, connected to Wi-Fi.
* Prusaslicer: Choose something to print then press G button (send to printer)
* First attempt times out (10000ms)
* Second attempt succeeds immediately

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Debian Trixie

### Printer model

Mini+

murk-sy, 2023-08-03:
Duplicate/See also https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2772

doegox, 2023-08-03:
Not sure it's related. I've never experienced a loss of Wi-Fi and I don't need to restart the printer. I just need to start the remote job twice from the pc

murk-sy, 2023-08-03:
Try opening the prusalink page before sending gcode next time. It's possible that it just temporarily loses connection, though apparently it should attempt to reconnect once every 60 seconds. However, if prusalink webpage opens normally and prusaslicer still times out the first time, it's likely a different problem.

doegox, 2023-08-03:
I don't use prusalink webpage at all, I just use prusaslicer and its G button to send to printer

murk-sy, 2023-08-03:
This is just to verify if the printer has network connection. If the webpage is accessible, it has network connection and should work immediately. But if it doesn't, it needs further digging to investigate.

doegox, 2023-08-03:
ha ok I can't test for the next weeks, will do when back.

closed 2026-09-29 by prusaslicer-bot[bot]