prusa3d/PrusaSlicer #11042: Step file has holes

forReason, 2023-07-26:
### Description of the bug

I created a mediocre complex model in solidworks. Exported with Step AP214 and AP 203.
The Model has holes in prusaslicer but when opening with Solidworks, its fine.
[image]

the exported stl file is also fine.

### Project file & How to reproduce

[bug report.zip](https://github.com/prusa3d/PrusaSlicer/files/12169045/bug.report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10, Ubuntu

### Printer model

-

forReason, 2023-07-26:
the zip file contains the slicer project +3d models

foreachthing, 2023-07-26:
You really should avoid fillets running into a single point.
I took the liberty and "update" your model. I just made it a bit wider that the dimples don't run into the fillet. Just have a look. You could also remove the top fillet all together as it's the culprit to your issue. I don't think it's the slicer's problem - this is a STEP import problem and trust me, PS does sometimes a better job than our FEA software...

[IKEA Ivar E3D Revo Nozzle block.zip](https://github.com/prusa3d/PrusaSlicer/files/12171677/IKEA.Ivar.E3D.Revo.Nozzle.block.zip)

Note to SolidWorks:
Make sure your sketches are fully defined - this will save you lots of troubles later when you need to adjust some dimensions ;-)

forReason, 2023-07-26:
Hi there,

Thanks for your insights on the model, I appreciate your insights and time you invested into this. While I see your point about fillets running into a single point, this aspect is intentional for aesthetics. Similarly, removing the top fillet would introduce a sharp, undesirable edge, which would lead to manual work after printing the file.

Interestingly, online STEP viewers interpret the file correctly, implying that the issue could be with PrusaSlicer's interpretation. Also, the STL prints correctly, suggesting the STEP file design isn't the root problem.
I understand this might be a minor edgecase which might not justify putting a lot of effort chasing down and fixing. After all, I have a workable workaround with exporting the file as stl.

While I thank you for the input regarding fully defining sketches in SolidWorks. I understand this is important and work to improve my Modeling skills.

Thanks again for your suggestions and help on this project.

Best,

Julian

foreachthing, 2023-07-26:
Maybe make your fillet just a riny bit larger to avoid running into a point... maybe!

splatteredbits, 2026-09-13:
I'm also having this problem. I have a model where several body planes touch. I export them as a single body, and Prusa Slicer is for some reason treating them as separate. Here's the body/model, from a .3mf:

[image]

And here's the sliced preview. As you can see, Prusa doesn't join the parts together but instead is using a bridge infill:

[image]