prusa3d/PrusaSlicer #11043: Add complete individual objects on the first layer option

Jamesgt, 2023-07-26:
**Is your feature request related to a problem? Please describe.**
When printing multiple objects bed adhesion is not the best if only the perimeters are printed first for all then solid infills again for all. You already have "Complete individual objects" option, but that is for the full height.

**Describe the solution you'd like**
I would prefer to have this option only for the first layer as I think I'll get better results on MK3S+.

**Describe how it would work**
An extra option in print settings that would be "Complete individual objects on the first layer".

**Describe alternatives you've considered**
I can print parts one by one, but that is not really an alternative for all night printing.

foreachthing, 2023-07-27:
I am currently working on a post processing script to this "issue". I'll post my prototype as soon as it's ready...

MadTooler, 2024-04-22:
> So, I think I got it. To be able to print the first layer first, clone this: https://github.com/foreachthing/Slic3rPostProcessing/tree/dev to some place on your computer (make sure it's on _dev_ branch!). Best is a path with no spaces! Then use it as normal post-processing script (see [readme](https://github.com/foreachthing/Slic3rPostProcessing/tree/dev#installation)) with these parameters: `--iob` and `--iobfl`.
> 
> Also, add this line to START_GCODE in machine settings: `; first_layer_height = {first_layer_height}`
> 
> The script has currently been tested with PrusaSlicer 2.6 and OrcaSlicer 1.6.3.
> 
> ## PLEASE CHECK YOUR GCODE BEFORE SENDING IT TO THE PRINTER!!!

@foreachthing I have been trying to use your PP to achieve the first layer for all then sequential without luck.  The closest I get is   to have it run on the output file, but with "Error code: 1" result during end of export.

I am running win10 64 with PS 2.7.4 and Python 3.12 64b

I have added the START_GCODE line and have my output set as:
fullpath\python.exe fullpath\Slic3rPostProcessing-dev\main.py --iob --iobfl;

If I remove the --iob --iobfl it runs without error, but nothing is done.  I also tried not including those and a dummy of --fl and I did get any error from your script that it was not valid.

Any idea if I am doing something wrong or if the newer versions are keeping your PP from working?

Thanks

MadTooler, 2026-09-18:
I still find this an important and needed feature for sequential printing.

foreachthing, 2026-10-28:
Yeah, maybe one can vibecode something like this...
I'm no programmer and have lost intrest after 1000000 failed tries. Maybe some day... 😏