prusa3d/PrusaSlicer #11045: Allow negative elephant foot comensation

Matszwe02, 2023-07-26:
**Is your feature request related to a problem? Please describe.**
When I make multicolor prints sequentially (like voron has in its multicolor bodies) I print the color first, then print over it with base color. I want the main color to have a standard elephant foot compensation, but there will remain a small distance between the colors. I want the inside part to have negative elephant foot compensation to extend a bit.

**Describe the solution you'd like**
Allow negative elephant foot values

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

Bumblebiber, 2023-07-28:
Why not just go closer to the bed or increase flow rate?

Matszwe02, 2023-07-30:
> Why not just go closer to the bed or increase flow rate?

First of all, my first layer flowrate os ok, when I lower it, that much flowrate is horrible. Second of all, materials like ABS tend to turn white at areas where it's squished into the bed

closed 2026-09-29 by prusaslicer-bot[bot]