prusa3d/PrusaSlicer #11048: PrusaSlicer Updated. Error

sslang, 2023-07-27:
### Description of the bug

Just updated the Slicer and get this error on exporting to USB. Suggestions please

Failed processing of the output_filename_format template.
Parsing error at line 1: Non-integer index is not allowed to address a vector variable.
{input_filename_base}_{nozzle_diameter[initial_tool]}n_{layer_height}mm_{printing_filament_types}_{printer_model}_{print_time}.gcode

Steve@chilternweb.com                                       

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

HP Laserjet

arirangz, 2023-08-02:
Hello
Same issue with MK3S on windows with same version of PrusaSlicer
Thanks

ulope, 2023-08-04:
Just got the same error for the first time.
I've been using 2.6 (and prior to that the 2.6 RCs) for weeks without encountering this.

A restart seems to "fix" it.

Version: PrusaSlicer-2.6.0+MacOS-arm64-202306191404
System: macOS 13.4 x86_64

Khalinda, 2023-08-23:
I'm on Windows 11.  I've had 2.6.0 installed for 2 or 3 weeks and this problem just started happening today.  A restart doesn't fix the problem.  I uninstalled 2.6.0 and reinstalled 2.5.2.  That fixes the problem, but I lost all of my settings.

nejec, 2023-08-30:
Slicing before exporting to GCode fixed the issue for me.

closed 2026-09-29 by prusaslicer-bot[bot]