prusa3d/PrusaSlicer #11050: Post-processing script not found when project or stl is opened from browser or recent files in task bar right click menu.

gstrabeli, 2023-07-27:
### Description of the bug

**Issue:** When double clicking an STL or .3mf file from the browser, prompting Prusa Slicer to open, the error "The configured post-processing script does not exist: ArcWelder.exe" comes up when trying to export the G-code. The same thing happens when selecting a recent file in the context menu when right clicking the program icon in the task bar

**Work around:** Save the project, close/reopen Prusa Slicer and reload it from the File menu.

**How to avoid it:** Opening Prusa Slicer first, then loading the project or importing an STL from the File menu.

### Project file & How to reproduce

Download ArcWelder: https://github.com/FormerLurker/ArcWelderLib
Place ArcWelder.exe in the main folder of Prusa Slicer, ie: C:\Program Files\Prusa3D\PrusaSlicer\
Add "ArcWelder.exe;" (minus the quotes) to Print Settings > Output options > Post-processing scripts 
(Error still happens if you put the full path to the file)
Save the profile and close Prusa Slicer.

**First way:** From the browser go to a project or stl and double click it to have Prusa Slicer open and load it. Slice and export the G-code to receive the error. 

**Second way:** Right click the Prusa Slicer icon from the task bar and choose a recent file to have Prusa Slicer open and load it. Slice and export the G-code to receive the error. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

MK3 S+

foreachthing, 2023-07-27:
In Windows, you need quotes around the whole path, if you have spaces in it. Best would be to put arcwelder somewhere else. I.e. c:\scripts\arcwelder.

gerusz, 2024-05-16:
Problem persists in PrusaSlicer 2.7.4, reproduced while launching PrusaSlicer from Printables. Workaround still works.

Possible cause: working directory could be set to the model's directory when the application is used to open a model, messing up all paths? Just guessing here.

closed 2026-09-29 by prusaslicer-bot[bot]