prusa3d/PrusaSlicer #11051: Prusaslicer doesn't open any more on MacOS Sonoma (beta4)

jedisct1, 2023-07-27:
### Description of the bug

On MacOS Sonoma, Prusaslicer 2.6.0 loads, but then the application window doesn't show up any more.

Trying to run it from the command line prints:

```
❯ ./PrusaSlicer
[2023-07-27 23:12:44.769248] [0x00000001e67dfec0] [trace]   Initializing StaticPrintConfigs
[2023-07-27 23:12:45.248774] [0x00000001e67dfec0] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file, not in dyld cache)
```

### Project file & How to reproduce

Install the current Sonoma version.

Install Prusaslicer.

Start it :)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

macOS 14.0

### Printer model

MK3S, MK4 and XL, but irrelevant.

jedisct1, 2023-07-27:
Fixed by disabling "restore position on start".

closed 2023-07-27 by jedisct1

neilyoung, 2023-10-29:
Could you please elaborate, where you did disable this? Having the same issue

jedisct1, 2023-10-29:
@neilyoung Configuration > Preferences > General > Restore window position on start