prusa3d/PrusaSlicer #11055: Force Quite/Crash on export

All4Off, 2023-07-29:
### Description of the bug

Open STL or repaired OBJ (same file), simplify model to any reduction setting, export plate as STL/OBJ, name file... when committing this export with return key, 2.6.0 (latest release) force quits without an export.

### Project file & How to reproduce

See above

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Mac OS v13.5

### Printer model

(I think you mean to say "Please, fill this in when it seems RELEVANT.")

BBC-Wheel, 2023-07-30:
PrusaSlicer STL export crashing on MacOS 13.5 : Same issue !  Here is more info (=>crash report).

Process:               PrusaSlicer [1150]
Path:                  /Volumes/VOLUME/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0+MacOS-x64-202306191415 (PrusaSlicer-2.6.0+MacOS-x64-202306191415)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-07-30 09:14:24.7011 +0200
OS Version:            macOS 13.5 (22G74)
Report Version:        12
Anonymous UUID:        42E68BE1-835E-A19D-033F-B127A15C87D9

Time Awake Since Boot: 550 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000001931917fe8
Exception Codes:       0x0000000000000002, 0x0000001931917fe8

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [1150]

VM Region Info: 0x1931917fe8 is in 0x1000000000-0x7000000000;  bytes after start: 39486324712  bytes before end: 372830535703
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)         fc0000000-1000000000   [  1.0G] ---/--- SM=NUL  ...(unallocated)
--->  GPU Carveout (reserved)    1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
      GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Scrazzled, 2023-08-23:
Also occurs on Win 11 Pro. Crashes when attempting to export STL. 
Faulting module path: C:\Program Files\Prusa3D\PrusaSlicer\PrusaSlicer.dll
Exception code: 0xc0000005
Faulting application name: prusa-slicer.exe, version: 2.6.0.0

PhilippMolitor, 2023-09-07:
Happens to me too, STLs where it fails where 180-290MB in size. smaller files seem to work without issues. Crashes without warning, immediately after clicking "save" when exporting after simplifying. 

OS: macOS 13.5 (Intel)
PrusaSlicer: 2.6.0

lukasmatena, 2023-09-08:
Yes, this issue was fixed in 2.6.1-rc2. This one and #11188 are duplicates. Closing.

closed 2023-09-08 by lukasmatena