prusa3d/PrusaSlicer #11056: Solid Infill Threshold Area not working

bbnaz12, 2023-07-29:
### Description of the bug

Updated to Prusaslicer 2.6.0 and the solid infill threshold area in filling in small areas no matter the value input in the field.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

MK3S+

Jan-Soustruznik, 2023-08-02:
Hello, @bbnaz12 , 

Thank you for your report, could you share the .stl or .3mf file, please? 

thank you have a nice day

bbnaz12, 2023-08-02:
[CrystalDragon_NewLegs_Curled_0.2mm_PLA_MK3S_13h0m.zip](https://github.com/prusa3d/PrusaSlicer/files/12240393/CrystalDragon_NewLegs_Curled_0.2mm_PLA_MK3S_13h0m.zip)

Jan-Soustruznik, 2023-08-04:
Issue closed as duplicate #10607

closed 2023-08-04 by Jan-Soustruznik