prusa3d/PrusaSlicer #11058: Use arachne for solid infill

Matszwe02, 2023-07-30:
**Is your feature request related to a problem? Please describe.**
There are many models that, when infill angle set to 0°, the top solid infill ends too soon, thus creating a visible gap.
[image]

**Describe the solution you'd like**
Add thinner lines to the last part of the model. Maybe even apply the line width modifier to a few lines back, so it's covered more evenly.

**Describe how it would work**
Overcompensate for solid infill, by rather filling too much space than creating gaps. Then apply arachne to these lines that will be squished.

**Describe alternatives you've considered**
- Filling at 45°. It hides the issue, not eliminates it, creates (personally) worse top surface, still prone to issues when models have 45° faces.
- ironing - due to it being applied over the whole top layer, it's filling a bit more of it, and letting the plastic flow to these holes.

**Additional context**
Add any other context or paint drawing about the feature request here.

Jan-Soustruznik, 2023-08-01:
Hello, @Matszwe02 , 

thank you for your feedback, could you share some models where the problem is occurred, please?

Have a nice day.

Matszwe02, 2023-08-01:
Sure!
[[a]_Guidler_x1.zip](https://github.com/prusa3d/PrusaSlicer/files/12230453/a._Guidler_x1.zip)

Layer 65 and 135

closed 2026-09-29 by prusaslicer-bot[bot]