prusa3d/PrusaSlicer #11060: Segmentation fault when running with "--help-fff" or "--help-sla" options

JohnnyB83, 2023-07-31:
### Description of the bug

Version
PrusaSlicer-2.6.0+linux-x64

Operating system type + version
Debian GNU/Linux 11 (bullseye)
3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+ (firmware version not applicable to problem)

Behavior
Run prusa-slicer with either the ---help-fff or ---help-sla options in the command line
Once the Misc options section is displayed on the screen, a segmentation fault occurs and prusa-slicer exits
Error message in console after running the command: ```qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault```

### Project file & How to reproduce

How to reproduce:
Run `prusa-slicer --help-fff` as a CLI command

[config.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/12210725/config.ini.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+linux-x64

### Operating system

Debian GNU/Linux 11 (bullseye)

### Printer model

Original Prusa i3 MK3S+

combolek, 2023-09-02:
This is also reproducible in the latest master. I submitted PR #11234 to fix it.

zenda97, 2023-09-04:
Reproduced, thank you for the report.
SPE-1859

bubnikv, 2023-09-05:
Thanks, merged. It will be fixed in the 2.6.1 final.

closed 2023-09-05 by bubnikv

mkoistinen, 2023-09-10:
Looks like this also fixed issue #10950. Thank you.