prusa3d/PrusaSlicer #11062: Opening project file through explorer or Prusaslicer interface gives different results

verne64, 2023-07-31:
### Description of the bug

I've saved a projectfile, but when it's opened by double clicking the file in windows explorer, it doesn't load the profiles and settings I have saved, also the location of my meshes is off. When I open the same projectfile through the prusaslicer interface (file>open project... or file>recent projects) it does load the profiles and settings I saved. I've attached screenshots of the Prusaslicer window as it appears after opening. The first one shows the expected result after opening through prusaslicer, the second one after opening through windows explorer. When opening through explorer I also get an error mentioning several objects are positioned at multiple heights, even though they are not.  This seems to happen for all 3MF files I have on my hard drive, which were all saved from prusaslicer.  
[image]
[image]
[image]

### Project file & How to reproduce

[Projectfile.zip](https://github.com/prusa3d/PrusaSlicer/files/12214697/Projectfile.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+Win64

### Operating system

Windows 11 22H2

### Printer model

Ultimaker 2 With Olsson Block

neophyl, 2023-07-31:
If I recall correctly this was done by user request.  When double clicking it in effect performs an Import of the file rather than an Open.  This is because it seems that users were getting stressed that opening a 3mf was replacing their profiles with the ones stored in the project.   It used to work how you want but user pressure got it changed.  

Personally I was on the leave it how it was but hey ho. To me double click means Open so it should load in everything but as I say the consensus of the vocal users was they wanted it to work as it does now.

verne64, 2023-07-31:
Thanks! I've read (most of) that topic now and I am amazed at the outcome. I completely disagree with the OP there and feel the project file should encompass all the project parameters and have built my workflow around this idea (especially useful in sharing work with less 3D-printer-savvy colleagues), but I guess that's a matter of preference.

But the idea of having different outcomes depending on the way you open a file? That is just a major mistake in UI design, if I've ever seen one. This just opens up a very very easy way to make mistakes in your workflow. Now whenever I share a project file with someone, I'm going to have to include an instruction on which way it needs to be opened!? That instruction is probably not going to be read, because everybody already knows how to open a file, right? So I will have to include a verbal reminder as well and then I need to come over and help them, when it's not working right because they ignored the verbal reminder as well. 

Opening a file should just always yield the same result, regardless of the way you open it. And in the case of a "project"file should encompass the "project" if you ask me, but I guess that's open for debate.

neophyl, 2023-07-31:
Entirely agree.  Which is why I tried to weigh in on the original issue. Didn’t make sense then and doesn’t now but you can see what the decision was. 
I’ve basically had to change my workflow to always start PRUSA slicer first on its own and then use file>open at all times now.

Area5142, 2023-08-01:
Totally agree too.
Opening a project file should open the project whit all settings in the project, not just import the models to what ever was last used.
That explains why I sometimes get a wrong setting when opening a project...

closed 2026-09-29 by prusaslicer-bot[bot]