prusa3d/PrusaSlicer #11063: Seams on first Layers in Vase Mode?

aclucca, 2023-07-31:
### Description of the bug

in Vase Mode should not be seams, but there are some in the first two layers?

### Project file & How to reproduce

[vase.zip](https://github.com/prusa3d/PrusaSlicer/files/12222335/vase.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 10 64bit

### Printer model

Prusa MK4

murk-sy, 2023-08-03:
Appears to be affected by `Skirt Height`, despite number of loops being 0.

0 bottom solid layers, 3 skirt height
[image]

0 bottom solid layers, 9 skirt height
[image]

0 bottom solid layers, 0 skirt height
[image]

Keep in mind 0 bottom solid layers was never really intended to be used (or actually tested) and will probably your build plate - it's best if you set skirt height to 1 so you have 1 solidly attached layer printed at a normal layer height.

Non-MK4 users: Keep in mind also to check if `Elephant foot compensation` will cause issues with this.

aclucca, 2023-08-03:
> Appears to be affected by `Skirt Height`, despite number of loops being 0.
> 
> 0 bottom solid layers, 3 skirt height [image]
> 
> 0 bottom solid layers, 9 skirt height [image]
> 
> 0 bottom solid layers, 0 skirt height [image]
> 
> Keep in mind 0 bottom solid layers was never really intended to be used (or actually tested) and will probably destroy your build plate - it's best if you set skirt height to at least 1 so you have 1 solidly attached layer printed at a normal layer height.
> 
> Non-MK4 users: Keep in mind also to check if `Elephant foot compensation` will cause issues with this.

perfect - problem solved with this workaround :-)

so the bug is, nevertheless 0 Skirt Loops defined, Skirt Height isn´t ignored

closed 2026-09-29 by prusaslicer-bot[bot]