prusa3d/PrusaSlicer #11064: Prusa Change Color not work

cpeque, 2023-07-31:
### Description of the bug

In Prusa when the color changes, the extruder takes time to send filament. It generates an empty layer and the objects are separated. The same object on the same printer with Cura works ok. This happens to me with an Ender 3 - Neo Printer

### Project file & How to reproduce

[Prusa_BAD_Robert_Secador_v2_X5_1h7m_0,20mm_2,29m_PLA_ENDER3NEO.zip](https://github.com/prusa3d/PrusaSlicer/files/12222990/Prusa_BAD_Robert_Secador_v2_X5_1h7m_0.20mm_2.29m_PLA_ENDER3NEO.zip)
[2023-07-31 20_51_33-WhatsApp.zip](https://github.com/prusa3d/PrusaSlicer/files/12223002/2023-07-31.20_51_33-WhatsApp.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Versión 2.6.0+win64

### Operating system

Windows 11 Pro 64-biy

### Printer model

Creality Ender 3 Neo

cpeque, 2023-08-02:
[image]

rtyr, 2024-02-18:
Color change behavior is mostly handled by the firmware, but you can try PS2.7.2-alpha2, where we improved the placement of the color change command.

closed 2024-02-18 by rtyr