prusa3d/PrusaSlicer #11070: Crash on cut

andyylin, 2023-08-02:
### Description of the bug

I'm trying to cut this part at 1mm from the bottom, but PrusaSlicer crashes.

[image]

### Project file & How to reproduce

[tripod-universal-foldable-v3.zip](https://github.com/prusa3d/PrusaSlicer/files/12238789/tripod-universal-foldable-v3.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 11

### Printer model

MK3S+ with MMU2S, firmware 3.13.0 (3.0.0)

Jan-Soustruznik, 2023-08-02:
Hello, @andyylin , 

thank you for report. We will try to fix in some next release. 

issue created: SPE-1832

have a nice day

FidelCapo, 2023-08-14:
This issue should be fixed and the fix will be part of the next release. 
Closing.

closed 2023-08-14 by FidelCapo