prusa3d/PrusaSlicer #11074: PS 2.6.0: idle_temperature missing when extruder >3

TypQxQ, 2023-08-02:
### Description of the bug

When selecting a extruder higher than 3 for an object and slicing, idle_temperature[initial_extruder]} in custom G-code template start_gcode and {idle_temperature[next_extruder]} in custom G-code template toolchange_gcode causes:
Parsing error at line #: Trying to reference an undefined (nil) element of vector of optional values

### Project file & How to reproduce

[.45664.gcode.tmp.zip](https://github.com/prusa3d/PrusaSlicer/files/12242750/default.45664.gcode.tmp.zip)
[20mm_cube_20130605-1537-1he8n28-0 (1).zip](https://github.com/prusa3d/PrusaSlicer/files/12242763/20mm_cube_20130605-1537-1he8n28-0.1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0+win64-202306191221

### Operating system

Windows 11

### Printer model

Custom Klipper ToolChanger (qTC)

lukasmatena, 2023-08-03:
This is not a bug. PrusaSlicer is telling you that you asked for `{idle_temperature[initial_extruder]}`, but the respective extruder has that option unchecked (not used), meaning that there is no value that the placeholder parser can substitute for it. You have two ways out of it:
1. Enable idle_temperature in the profile and set it to the value that you want
2. Change the custom gcode to something like `{if is_nil(idle_temperature[initial_extruder])}M568 R{idle_temperature[initial_extruder]}{else}M568 R100{endif}`

TypQxQ, 2023-08-04:
Thank you! You are right!

closed 2023-08-04 by TypQxQ