prusa3d/PrusaSlicer #11075: Prusa XL Slicer 2.6.0

sith77, 2023-08-02:
### Description of the bug

When an object is using multimaterial painting and the object has selections that are colored to be used with multiple tools , if you use the cut feature to cut the object for any reason it removes all the painted sections and no longer has multi tool head setup . 

### Project file & How to reproduce

When an object is using multimaterial painting and the object has selections that are colored to be used with multiple tools , if you use the cut feature to cut the object for any reason it removes all the painted sections and no longer has multi tool head setup . 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Prusa XL

neophyl, 2023-08-08:
When you specify a surface for painting for something like support or colour etc it is part of a list of possible faces. It’s basically one big list and some of them are marked with whatever you have defined. If you cut the part, or even load it in again/replace it etc then the list is no longer valid so at the moment it resets everything.

Jan-Soustruznik, 2023-08-11:
Closed as duplicate

closed 2023-08-11 by Jan-Soustruznik