prusa3d/PrusaSlicer #11076: PrusaSlicer fails to upload to PrusaConnect (often)

BaileyMillerSSI, 2023-08-03:
### Description of the bug

[image]

As of recently I have been unable to reliably upload files to PrusaConnect using PrusaSlicer. The upload will get to 100% and then hang for a minute and then present this error.

### Project file & How to reproduce

[Shark Head.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12246144/Shark.Head.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+MacOS-arm64

### Operating system

macOS Ventura 13.5 (22G74)

### Printer model

Prusa Mk4

drchslrk, 2023-08-21:
Same here! This issue is very annoying and it appears randomly.

BaileyMillerSSI, 2023-08-21:
> Same here! This issue is very annoying and it appears randomly.

I have completely given up on uploading through the slicer. I have to export to file, then upload using the website.

drchslrk, 2023-08-25:
I've noticed that the upload works if the name of the physical printer, "Descriptive name for the printer," is changed beforehand.

1. Printer Settings 
2. Edit physical printer 
3. change the name -> ok 
4. select a random config 
5. now you see the new renamed config -> select it 
6. upload to Connect

I have to do this ervery time I restart the PrusaSlicer.

marvelgj, 2023-09-24:
Same here!
Uploads to Klipper stop at 6% with SuperSlicer (v2.5.59.2).

MacOS Ventura 13.5 - arm64

Jan-Soustruznik, 2023-09-26:
@BaileyMillerSSI , 

Thank you for you report, but problem is caused by connect service. We cannot to fix. We informed Connect team about this issue, they will try to fix is some next releases. 

Issue: CONNECT-2807

Issue closed

toddanglin, 2023-11-05:
Experiencing the same issue with PrusaSlicer 2.7.0 alpha (and a Prusa XL, FWIW). Does not appear it has been resolved yet, and like others have shared, it completely breaks the "Upload and Print" flow within PrusaSlicer.

While I can appreciate this is not strictly the Slicer's problem, it is frustrating that the Slicer team and the Connect team do not seem to be working more closely to resolve. Part of the appeal of integrated ecosystems (PrusaSlicer + PrusaConnect + Prusa printer) should be that the pieces work together and troubleshooting doesn't feel like working with multiple, disconnected companies. :/

In the meantime, is there any place to monitor/track the Connect issues?

johnboiles, 2023-11-08:
Also seeing the same thing with PrusaSlicer 2.6.1+MacOS+arm64 for the last couple months. PrusaSlicer is otherwise incredible, it just doesn't work with PrusaConnect

lukasmatena, 2023-11-11:
We noticed that this issue is only reported from macOS, which is suspicious. We have a hypothesis that it might in fact be caused by a bug in libCURL library shipped with macOS.

Could you please try [this build](https://slicerbuilds.prusa3d.com/lm/PrusaSlicer-2.7.0-beta1+1-lm-macos-libcurl-MacOS-universal-g57017e5da-202311111352.dmg) for us and let us know the result?  The build is 2.7.0-beta1 with a single change that would bypass the bug. We cannot reproduce the issue reliably, unlike some of you. Thanks.

yasars, 2023-11-11:
2.7.0 beta 1 same error ... I use only prusa link... Connect still 100% freeze and then same error..

toddanglin, 2023-11-11:
So far, I'm 2 for 2 successful uploads from PrusaSlicer to Connect with 2.7.0 beta 1 build.

I'll do more uploads with this new build and report back if I re-encounter the upload errors. Even with the 2.7.0 alpha build, uploads would sometimes work, so unclear as of yet if this has fully addressed the root cause. But good start!

For added debugging context, my environment:
- macOS 14.0
- Sending to Prusa XL 5 Tool via PrusaConnect

lukasmatena, 2023-11-11:
> So far, I'm 2 for 2 successful uploads from PrusaSlicer to Connect with 2.7.0 beta 1 build.

There is no bugfix relevant to this issue in 2.7.0-beta1, I would expect no difference between that and 2.7.0-alpha1 or 2.6.1. It is the **2.7.0-beta1+1-lm-macos-libcurl** build in which I hope this might be fixed.

yasars, 2023-11-12:
It also works from my side. I have already uploaded a file more than 10 times to the mk4 with the alpha firmware. It works well, the only thing I noticed is that it waits a little longer at 100% (a few seconds) but then it uploads ....the main thing is that it works again.. please include it in the next betas/aplhas/stables :) thanks

toddanglin, 2023-11-12:
> There is no bugfix relevant to this issue in 2.7.0-beta1, I would expect no difference between that and 2.7.0-alpha1 or 2.6.1. It is the **2.7.0-beta1+1-lm-macos-libcurl** build in which I hope this might be fixed.

Sorry, should have been more precise. I am testing with the 2.70-beta1 libcurl build you provided in this thread. And the Prusa XL is on the alpha firmware (FWIW).

I'm also now 4 for 4 with an experience similar to @yasars: there is a bit of a pause at 100% upload, but then it successfully completes. Thanks again for reopening this issue and finding a potential solution for macOS!

johnboiles, 2023-11-12:
I installed the `2.7.0-beta1+1-lm-macos-libcurl-MacOS-arm64` build. My work printer uses Prusa Connect so I'll evaluate the Prusa Connect issue over the next week.

I just saw another issue which is probably a fluke but _might_ be related to the libcurl change. My first download from printables.com when I hit `Open in PrusaSlicer` was excruciatingly slow (it just took ~ 2 mins for a 6MB file). Downloading the same file from the web browser was pretty much instantaneous (my speedtest.net results are ~370mbps down so it should take ~130ms). I haven't been able to repro again, so maybe just a weird internet fluke, but mentioning here in case other people see anything similar.

lukasmatena, 2023-11-13:
@BaileyMillerSSI @drchslrk @barn53
What about you, guys? Are you able and willing to test the experimental build I linked in https://github.com/prusa3d/PrusaSlicer/issues/11076#issuecomment-1806837778 ? I would like to be sure that it solves the issue. Thanks.

BaileyMillerSSI, 2023-11-13:
I'll see if I can give it a try this week. Tbh I e switched mainly to using my Bambu printer because it "just works". My Mk4 has been sitting turned off for weeks

SmithyAT, 2023-11-13:
@lukasmatena I can confirm that it works on my side. Was not able to upload with 2.6.1, your version worked immediately. I am on macOS 14.1.1
I guess you found a solution for it! Thanks a lot!

johnboiles, 2023-11-13:
I've sent three files to the printer (MK3S with Pi Zero + PrusaLink) via PrusaConnect today all without issue.

I did get one crash on this build though in `Slic3r::GUI::HighlighterForWx::invalidate`. Seems unlikely to be the libcurl change, should I report it somewhere?

lukasmatena, 2023-11-14:
@johnboiles Thanks. That really sounds to be unrelated. Could you create a separate issue for it and include steps to reproduce it and the backtrace?

toddanglin, 2023-11-14:
Reporting back...I'm now 6 for 6 successful uploading to Connect from this build of PrusaSlicer. No failures. It seems you have, at a minimum, found a common driver of the upload problems we were encountering on macOS. Thanks!

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1).
The problem was possibly caused by a bug in libCURL library shipped with the affected macOS versions (possibly https://github.com/curl/curl/issues/11353). We worked around the problem by using HTTP/1.1 for libCURL requests.

commit 4f18b6d0 (2023-11-16): Ask for HTTP/1.1 for CURL requests, hopefully will fix #11076. The problem is
apparently caused by https://github.com/curl/curl/issues/11353, requesting
HTTP/1.1 is a workaround. (On macOS, system libCURL is used.)
closed 2023-11-16 by lukasmatena