prusa3d/PrusaSlicer #11081: What is wrong with the "wipe" feature it wipes after seam point than drives back?

Doprintityourself, 2023-08-05:
### Description of the bug

From my point of view wipe is wiping from the wrong side
I am used that before a Retract or layer change occurs the wipe amount is BEFORE it occurs
Just have printed a part with 2.6 and switched the feature on and recognized a wired seam after the retract seem
checked the slicing and I am wondering if there is something going majorly wrong!?
Since when is the wipe after the layer change
See the video the part is printing anti clockwise but the wipe is after so it prints over the seem point and than wipes back to it !!!

https://github.com/prusa3d/PrusaSlicer/assets/70262325/f7d2c555-6ada-490a-a724-f1e062d708d9

This strange behavior is seen all over the part, so it is doing it generally wrong!?

### Project file & How to reproduce

Just enable wipe and slice a part with 2.6
(reproduced it also with 2.5.1 having laying arround)
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

Windows 10

### Printer model

TTSPLUS

Jan-Soustruznik, 2023-08-07:
Hello, @Doprintityourself , 

Could you share your project file(3mf), please? 

thank you, have a nice day

Doprintityourself, 2023-08-07:
Hello Jan  , 
here it is.
You can see that it de retracts and than wipes after before going into travel
[image]
the result is bad surface and strange artifacts like this:
[image]
Thanks for investigating!

[Wipe After Retract not before TTSPlus1.zip](https://github.com/prusa3d/PrusaSlicer/files/12274069/Wipe.After.Retract.not.before.TTSPlus1.zip)

closed 2026-09-29 by prusaslicer-bot[bot]