prusa3d/PrusaSlicer #11082: no perimeters when one perimeter

h4lf, 2023-08-05:
### Description of the bug

with some settings there is no perimeter (Arachne).

[image]
[image]
[image]

### Project file & How to reproduce

[Форма-Куб.zip](https://github.com/prusa3d/PrusaSlicer/files/12267084/-.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 10 22H2  19045.3208

### Printer model

Ender 3

Jan-Soustruznik, 2023-08-10:
Hello, @h4lf , 

thank you for your report. We will try to fix in some of new release. 

issue created: SPE-1837

Have a nice day

dougisfunny, 2023-10-24:
My issue was definitely a duplicate of this bug. But I did add the info that it is specifically due to the ratio of extrusion width on perimeter vs external perimeters that isn't specifically mentioned in this issue.

lukasmatena, 2023-10-24:
@dougisfunny We are aware of what causes it, although you are correct that it is not mentioned explicitly in this issue.

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena