prusa3d/PrusaSlicer #11083: bridges issue

real-artswan, 2023-08-06:
### Description of the bug

Bridges are printing totally wrong
[image]
one can expect of a way a filament pushed this way - whole the print into the trash, everything in the air.
With the "external perimeters first":
[image]

I had to put bridges speed to 10 (as a side effect I've got 10mm/s on top of infill what is annoying while they are not the "bridge" really)

### Project file & How to reproduce

[prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/12270621/prusa.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+linux-x64-GTK3 

### Operating system

linux 

### Printer model

voron 0.1

real-artswan, 2023-08-08:
Is there a way to separate speed of real bridges and spped of bridges on top of infill? May be a distance of bridge... would be very useful

Jan-Soustruznik, 2023-08-09:
Hello, @real-artswan 

I think that I can't give you a satisfy answer. With your setup can't be your model successfully printed. I think that necessary to have a supports or print the models separately or maybe try rotate model. Currectly in slicer doesnt exist any feature that may help to print your model as is it.

1. Feature "External perimeters first" may help with dimensional accuracy, but not with better logic of generate external perimeters.
2. Slowly printing of infill is caused by The Pressure equalizer smooths sudden changes in speed between two features (e.g. between printing infill and perimeters). More you can find there about this feature: https://help.prusa3d.com/article/pressure-equalizer_331504 

Issue closed together with https://github.com/prusa3d/PrusaSlicer/issues/11093

closed 2023-08-09 by Jan-Soustruznik

real-artswan, 2023-08-09:
And a separated option for bridge infilll speed is an option?

real-artswan, 2023-08-09:
or better a smarter detection of a bridge length... don't know. Why it prints infill like bridges?