prusa3d/PrusaSlicer #11086: Color change dollop problem

chrispmcgee, 2023-08-07:
### Description of the bug

This problem arises when printing an item with a small surface area piece, and that small surface area is the last thing to be printed before the color change. After swapping filaments, the nozzle moves back to its last known location, that small surface piece, and "touches" it. In so doing, it leaves a small dollop of the second color's filament on it. Then, when the nozzle gets around to printing around that area, it will hit that now-solidified dollop, knocking it and the layers before it off the bed.

I would expect that, after changing filament colors, the nozzle would go to the next layer's starting position and begin printing there rather than touching the previous layer's ending position first.

Here is the problem area. The middle piece of the "a" is the last thing to print with red filament:
[image]

After changing to a white filament, the nozzle returns to the last position, which is that middle "a" piece, and touches it, leaving a blob of white on it:
[image]

That blob of white gets knocked off when the nozzle sweeps over that spot, taking the red with it. The result is that the "a" no longer has a middle piece:
[image]

### Project file & How to reproduce

[Color-change-dollop-problem.zip](https://github.com/prusa3d/PrusaSlicer/files/12282148/Color-change-dollop-problem.zip)

1. Just print as normal, doing a color change when indicated.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 11 21H2, Build 22000.2176

### Printer model

Prusa i3 MK3S+

neophyl, 2023-08-07:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4432.

Jan-Soustruznik, 2023-08-09:
Closed issue as duplicated

closed 2023-08-09 by Jan-Soustruznik