prusa3d/PrusaSlicer #11087: [feature request] Add supported profiles for Elegoo Neptune 4 and Neptune 4 Pro printers

luisillo26, 2023-08-07:
**Is your feature request related to a problem? Please describe.**
Elegoo recently released its Neptune 4 and 4 Pro bed slingers but there are no supported profiles in PrusaSlicer for those printers.

**Describe the solution you'd like**
My request is to add profiles for such Elegoo printers given that Elegoo doesn't seem to support PrusaSlicer since they offer their version or Cura slicer. There are some Elegoo printers profiles in PrusaSlicer but not for Neptune 4 and Neptune 4 Pro.

neophyl, 2023-08-07:
Non PRUSA printer profiles are supplied by volunteer users. Until someone makes one and submits it then it can’t be included. Btw there’s a separate GitHub repository for the profiles.

kydreth, 2023-08-26:
This is the top result on Google. It provides no one any help. Here's a shot for those that stumble upon it searching for prusaslicer and elegoo (4 / 4 Pro)

- PrusaSlicer Elegoo FDM/FFF printers are all available in PrusaSlicer, make sure to update to the newest config. Neptune 4 / 4 Pro have been released and will hopefully be available in the coming weeks. For now:
  - https://www.youtube.com/watch?v=_C44Xmz9Pws YouTuber first to announce that I've seen
    - His hosted version of the exported config: https://www.dropbox.com/scl/fo/ki5j8qor4lj4k6e5cwn7u/h?dl=0&rlkey=5ytcubh5bu18dsbhys5273aaz install at your own risk as it's a full export from their PrusaSlicer and not an optimized version from @rtyr 

For now others probably have PrusaSlicer installed and just need to update the software or update printers available which they can read: https://www.nikkoindustries.com/blogs/news/how-to-update-prusaslicer
  - If not, download PrusaSlicer: https://www.prusa3d.com/page/prusaslicer_424/
    
 If you're looking for the config code repo it's here: 
- https://github.com/prusa3d/PrusaSlicer-settings
  - Elegoo: https://github.com/prusa3d/PrusaSlicer-settings/tree/master/live/Elegoo
    - Last config update for printers was Jan 2023 introducing Neptune 3 / 3 Pro: https://github.com/prusa3d/PrusaSlicer-settings/commit/80689a6f8f98e50e89ea61863bd52849df1a0c2e by @rtyr who most likely would consolidate the latest 4 / 4 Pro config for the community. We're all super appreciative of all their efforts! When done then we all can enjoy just picking 4 / 4 Pro from inside PrusaSlicer Configuration Wizard > Other Vendors > Elegoo FFF list of available printers UI.

I recommend this issue be closed as it's not related to this repo.

Kurpanik, 2023-09-19:
I can't recommend the links above in the Youtube tutorial at all. And in the comments there you can also see why: After copping both files,  there are way to many important settings wrong, some by really large values (retraction 45 instead of 0.45)

kydreth, 2023-09-19:
I have been putting my Elegoo Neptune 4 Pro through its paces. What worked best for me was using the prusaslicer standard settings for the Neptune 3 pro. Then slowly updating them with tuning steps such as: 
- https://youtu.be/rGyVjqCD2Pk?si=hMIYdlSZPFlgVtft ignoring things that don't apply for the 4 Pro
- Then https://youtu.be/mCcP8dffwLk?si=gB17xR148OerO66_ however I suggest skipping the z-axis step if you can press down and they're already flat or square to the base with two equal objects like the two cassette tapes he uses.

I also have copied over some custom gcode from Elegoo cura.

Start G-Code
Cura Default Start G-Code
```gcode
G28 ;home
G92 E0 ;Reset Extruder
G1 Z4.0 F3000 ;Move Z Axis up
G92 E0 ;Reset Extruder
G1 X1.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X1.1 Y80.0 Z0.28 F1500.0 E10 ;Draw the first line
G1 X1.4 Y80.0 Z0.28 F5000.0 ;Move to side a little
G1 X1.4 Y20 Z0.28 F1500.0 E20 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
```

PrusaSlicer Default Start G-Code
```gcode
M413 S0 ; disable Power Loss Recovery
G90 ; use absolute coordinates
M83 ; extruder relative mode
SET_MATERIAL MATERIAL='{filament_type[initial_extruder]}'
M104 S120 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
M140 S[first_layer_bed_temperature] ; set final bed temp
G4 S10 ; allow partial nozzle warmup
G28 ; home all axis
;G29 ; run abl mesh
M420 S1 ; load mesh
G1 Z50 F240
G1 X2 Y10 F3000
M104 S[first_layer_temperature] ; set final nozzle temp
M190 S[first_layer_bed_temperature] ; wait for bed temp to stabilize
M109 S[first_layer_temperature] ; wait for nozzle temp to stabilize
G1 Z0.28 F240
G92 E0
G1 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0
```

PrusaSlicer Combined Start G-Code
N/A, just use the Elegoo Neptune 3 Pro settings

End G-Code
Cura Default End G-Code
```gcode
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F1600 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z2 ;Raise Z more
G90 ;Absolute positioning
G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
```

PrusaSlicer Default End G-Code
Combining some of the elegoo cura gcode, but not all the variables are present in prusaslicer
```gcode
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F600 ; Move print head up{endif}
G1 X5 Y{print_bed_max[1]*0.8} F{travel_speed*60} ; present print
{if max_layer_z < max_print_height-10}G1 Z{z_offset+min(max_layer_z+70, max_print_height-10)} F600 ; Move print head further up{endif}
{if max_layer_z < max_print_height*0.6}G1 Z{max_print_height*0.6} F600 ; Move print head further up{endif}
M140 S0 ; turn off heatbed
M104 S0 ; turn off temperature
M107 ; turn off fan
M84 X Y E ; disable motors
```

PrusaSlicer Combined End G-Code
Combine some of the gcode, but we can't use all the variables since they're not available
```gcode
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F600 ; Move print head up{endif}
G1 X5 Y{print_bed_max[1]*0.8} F{travel_speed*60} ; present print
{if max_layer_z < max_print_height-10}G1 Z{z_offset+min(max_layer_z+70, max_print_height-10)} F600 ; Move print head further up{endif}
{if max_layer_z < max_print_height*0.6}G1 Z{max_print_height*0.6} F600 ; Move print head further up{endif}
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
```

luisillo26, 2023-10-20:
I would close this issue since, now I know the changes should be part of the [prusa3d/PrusaSlicer-settings](https://github.com/prusa3d/PrusaSlicer-settings) repo

closed 2023-10-20 by luisillo26